hbldh/bleak #222: Dotnet: notify callbacks not happening sometimes

dhalbert, 2020-06-18:
* bleak version: 0.6.4
* Python version: 3.7.7
* Operating System: Windows 10 Version 2004 and also Windows Insider Preview builds
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

I have been having trouble with notify callbacks not happening sometimes. It seems to be related to which version of Windows 10 I'm using, and which Bluetooth adapter.

When the failure happens, no notify callbacks happen at all. Cycling Bluetooth on and off in Windows Settings doesn't help. Yet I have also seen it mysteriously start working after removing and replacing the adapter (but not all the time). I've instrumented the callback registration code in `dotnet/client.py`, and it seems to be doing what it it's supposed to.

I am testing with two different adapters:

1. (Fails) I have a generic Bluetooth 4.0 USB dongle, which identifies via USB as VID/PID 0a12/0001, "Cambridge Silicon Radio, Ltd", "Bluetooth Dongle (HCI mode)". In Windows it shows up in Device Manager as  "Generic Bluetooth Radio". This dongle works fine on a Windows 10 Version 1909 laptop just fine. But on Version 2004, and also recent Windows Insider Preview builds,  I see the notify callback problem.

2. (Works) I also have a Windows desktop machine with built-in Bluetooth. The adapter is an Intel Dual Band Wireless-AC 3160 mini-card. Device Manager shows it as "Intel (R) Wireless Bluetooth (R)". This machine is running Windows 10 Version 2004. Using the built-in adapter, I do _not_ see the notify callback problem.

Has anyone else seen this kind of notification callback failure?

dhalbert, 2020-06-18:
I found a problem description for 2004. Take this as a warning about updating to 2004 right now: https://docs.microsoft.com/en-us/windows/release-information/status-windows-10-2004

> **Difficulty connecting to more than one Bluetooth device**
> Realtek and Microsoft have found incompatibility issues with certain versions of drivers for Realtek Bluetooth radios and Windows 10, version 2004 (the Windows 10 May 2020 Update). Windows 10 devices with affected Realtek Bluetooth radio drivers might be unable to connect or pair with more than one Bluetooth device at a time after updating.
> 
> To safeguard your update experience, we have applied a compatibility hold on Windows 10 devices with affected Realtek Bluetooth radio drivers installed from being offered Windows 10, version 2004 or Windows Server, version 2004 until the driver has been updated.
> 
> Affected platforms:
> Client: Windows 10, version 2004
> Server: Windows Server, version 2004
> Next steps: Microsoft and Realtek are working on a resolution and will provide an update in an upcoming release.
> 
> Note We recommend that you do not attempt to manually update using the Update now button or the Media Creation Tool until this issue has been resolved.

_

hbldh, 2020-06-24:
Good thing to document, I think, but it does seems to be a hardware-related problem. I will leave this issue open until further notice.

chenyangvk, 2020-08-12:
same here. win 10 2004

dlech, 2022-07-25:
We dropped support of the pythonnet backend some time ago. If the issue still exists in the winrt backed, let's start a new issue.

closed 2022-07-25 by dlech