hbldh/bleak #228: Error During simple device scan

DionisisBozikis, 2020-06-26:
* bleak version: 0.6.4
* Python version:  3.7.5
* Operating System: Ubuntu 18.04.4 LTS"
* BlueZ version (`bluetoothctl -v`) in case of Linux:  5.48

### Description
I am trying to simply run the examples so i can get comfortable with the library. The error occurred when running the exact scan/discover example, but failed with the following error:

```
python3 ./python_scripts/bleak_discovery.py
Traceback (most recent call last):
  File "./python_scripts/bleak_discovery.py", line 10, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.7/asyncio/base_events.py", line 579, in run_until_complete
    return future.result()
  File "./python_scripts/bleak_discovery.py", line 5, in run
    devices = await discover()
  File "/home/user/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/discovery.py", line 174, in discover
    ).asFuture(loop)
txdbus.error.RemoteError: org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 2 matched rules; type="method_call", sender=":1.456" (uid=1000 pid=14370 comm="python3 ./python_scripts/bleak_discovery.py " label="unconfined") interface="org.freedesktop.DBus.ObjectManager" member="GetManagedObjects" error name="(unset)" requested_reply="0" destination="org.bluez" (uid=0 pid=25387 comm="/usr/lib/bluetooth/bluetoothd " label="unconfined")

```

DionisisBozikis, 2020-06-26:
solved by running the script with root permissions...

closed 2020-06-26 by DionisisBozikis

dlech, 2020-06-26:
Root permissions should not be needed. Make sure your user account is a member of the `bluetooth` group.

hbldh, 2020-06-26:
Don't run you python code as sudo. Do `sudo usermod -a -G bluetooth <your-username>` as @dlech advises.

If that still does not work, check this: https://github.com/sputnikdev/eclipse-smarthome-bluetooth-binding/issues/9 or this: https://unix.stackexchange.com/questions/348441/how-to-allow-non-root-systemd-service-to-use-dbus-for-ble-operation