hbldh/bleak #229: Device was not found.

darkzbaron, 2020-06-26:
* bleak version:0.6.4
* Python version: Python 3.6.0rc2 (v3.6.0rc2:800a67f7806d, Dec 16 2016, 23:22:07) [MSC v.1900 64 bit (AMD64)] on win32
* Operating System: Windows 10

### Description
I am trying to catch a "Device with address [ADDRESSMAC] was not found." and perform an action when this occurs. Ideally a retry or something like this.
Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.

### What I Did
I am using the example code but I can't find where the error would be stored and how to catch it with an async method

Any help would be more than welcome.

```
# -*- coding: utf-8 -*-
"""
Notifications
-------------

Example showing how to add notifications to a characteristic and handle the responses.

Updated on 2019-07-03 by hbldh 

"""

import logging
import asyncio
import platform

from bleak import BleakClient
from bleak import _logger as logger

CHARACTERISTIC_UUID = (
    "f000aa65-0451-4000-b000-000000000000"
)  # <--- Change to the characteristic you want to enable notifications from.

def notification_handler(sender, data):
    """Simple notification handler which prints the data received."""
    print("{0}: {1}".format(sender, data))

async def run(address, loop, debug=False):
    if debug:
        import sys

        # loop.set_debug(True)
        l = logging.getLogger("asyncio")
        l.setLevel(logging.DEBUG)
        h = logging.StreamHandler(sys.stdout)
        h.setLevel(logging.DEBUG)
        l.addHandler(h)
        logger.addHandler(h)

    async with BleakClient(address, loop=loop) as client:
        x = await client.is_connected()
        logger.info("Connected: {0}".format(x))

        await client.start_notify(CHARACTERISTIC_UUID, notification_handler)
        await asyncio.sleep(5.0, loop=loop)
        await client.stop_notify(CHARACTERISTIC_UUID)

if __name__ == "__main__":
    import os

    os.environ["PYTHONASYNCIODEBUG"] = str(1)
    address = (
        "24:71:89:cc:09:05"  # <--- Change to your device's address here if you are using Windows or Linux
        if platform.system() != "Darwin"
        else "243E23AE-4A99-406C-B317-18F1BD7B4CBE"  # <--- Change to your device's address here if you are using macOS
    )
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run(address, loop, True))
```

hbldh, 2020-06-30:
I will try to make a new example file for this shortly.

darkzbaron, 2020-06-30:
Hi, thanks so much for the work. I think the disconnect example doesn't work with my system too! Thanks for making the only working Python Bluetooth LE Api on Windows!

dlech, 2021-10-06:
This sounds like a use case for #527, so will close in favor of that issue.