hbldh/bleak #234: [macOS] AttributeError: 'CBCentralManager' object has no attribute 'isScanning'

mfilippo, 2020-06-30:
* bleak version: 0.6.4
* Python version: 3.7.0
* Operating System: macOS 10.12.6

### Description

I just wanted to run the basic scan:

```python
# scanner.py

import asyncio
from bleak import discover

async def run():
    devices = await discover()
    for d in devices:
        print(d)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```
### What I Did

From my virtual environment:

```
$ python scanner.py
Traceback (most recent call last):
  File "scanner.py", line 12, in <module>
    loop.run_until_complete(run())
  File "/usr/local/Cellar/python/3.7.0/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 568, in run_until_complete
    return future.result()
  File "scanner.py", line 6, in run
    devices = await discover()
  File "/Users/matteo/virtualenvs/bt-scanner/lib/python3.7/site-packages/bleak/backends/corebluetooth/discovery.py", line 35, in discover
    await cbapp.central_manager_delegate.scanForPeripherals_(scan_options)
  File "/Users/matteo/virtualenvs/bt-scanner/lib/python3.7/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 121, in scanForPeripherals_
    while self.central_manager.isScanning():
AttributeError: 'CBCentralManager' object has no attribute 'isScanning'
```

dlech, 2020-06-30:
`isScanning` was introduced in  macOS 10.13 so we will need a workaround for the call at https://github.com/hbldh/bleak/blob/release/v0.7.0/bleak/backends/corebluetooth/CentralManagerDelegate.py#L130 to support older macOS.

hbldh, 2020-06-30:
The last minute fix in version 0.7.0 might just do the trick for this one. @mfilippo please try it out and report back here if a more rigorous fix is needed.

closed 2020-06-30 by hbldh

mfilippo, 2020-07-02:
It's fixed. Thx!