hbldh/bleak #238: Method "Connect

Matt-Hicks-Bose, 2020-07-02:
* bleak version: 0.7.0
* Python version: 3.7.3
* Operating System: Raspbian GNU/Linux 10 (buster)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.

### What I Did

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here.
```

hbldh, 2020-07-02:
> Trying to connect to a non-existent BLE device results in an exception.

Well, shouldn't it throw an exception? Maybe a better one though, e.g. a BleakError.

It generally means that the BlueZ deamon does not have a device with that address stored in its list of discovered devices. Run `discover` for longer or set `timeout` higher in the `BleakClient`.

closed 2020-07-02 by hbldh

Matt-Hicks-Bose, 2020-07-03:
Thanks, that is a better error description and should reduce confusion.

commit d78d78d2 (2020-07-02): Improved error handling on BlueZ backend.

Fixes #238