hbldh/bleak #239: BlueZ error when connecting to BLE device

Matt-Hicks-Bose, 2020-07-02:
* bleak version: 0.7.0
* Python version: 3.7.3
* Operating System: Raspbian GNU/Linux 10 (buster)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

BlueZ error when connecting to BLE device. This is happening in my application code, but also with the `service_explorer.py` AND `get_services.py` examples. It does work <10% of the time, but it usually results in an exception.

### What I Did
```python
$ python3 service_explorer.py
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 151, in connect
    ).asFuture(self.loop)
txdbus.error.RemoteError: org.bluez.Error.Failed: Software caused connection abort

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "service_explorer.py", line 64, in <module>
    loop.run_until_complete(run(address, loop, True))
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "service_explorer.py", line 29, in run
    async with BleakClient(address, loop=loop) as client:
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/client.py", line 49, in __aenter__
    await self.connect()
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 154, in connect
    raise BleakError(str(e))
bleak.exc.BleakError: org.bluez.Error.Failed: Software caused connection abort
```

Matt-Hicks-Bose, 2020-07-02:
Maybe related to #238 ?

hbldh, 2020-07-02:
I bellieve that this is some error in BlueZ or your BLE device. Bleak just reports the OS level error. 

There isn't much I can do about this I am afraid. Try connecting to your device through `gatttool` or other BLueZ using means and try to see if that works.

dlech, 2020-07-02:
I find it useful to use Wireshark to capture Bluetooth packets to see what is actually going on.

Matt-Hicks-Bose, 2020-07-03:
I started digging into it with wireshark today, I will update when I know more.

RulerKun, 2020-07-07:
Also encountered this, What can I do to help in processing this error?

RulerKun, 2020-07-08:
is it possible that this is what happening on the background?

Since the first scan wasn't terminated if you fired the scan again, it will fail

[image]

hbldh, 2020-09-25:
The `bleak.exc.BleakError: org.bluez.Error.Failed: Software caused connection abort` seems to happen to many users. It might be alleviated by #236, or at least it might yield better error reports. Will close this, since there is no actionable content in this issue right now. Feel free to reopen if anyone has new information.

closed 2020-09-25 by hbldh

jaykishan-solanki, 2022-10-11:
DBUS BLE exception while bluetooth pairing
Exception in thread Thread-12:
Traceback (most recent call last):
File "/home/root/obd/obd_gen2_ble_py/blemodule/adapter.py", line 146, in powered
self.adapter_props.Set(constants.ADAPTER_INTERFACE, 'Powered',
File "/usr/lib/python3.8/site-packages/dbus/proxies.py", line 141, in call
return self._connection.call_blocking(self._named_service,
File "/usr/lib/python3.8/site-packages/dbus/connection.py", line 652, in call_blocking
reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name :1.1 was not provided by any .service files

can you plese guide me how to solve this error?and what is the root cause of this error

dlech, 2022-10-11:
This issue has been closed, so it would be better to start a new discussion at https://github.com/hbldh/bleak/discussions.

Be sure to include Bleak version, operating system, the source code and logs (see troubleshooting page of Bleak docs).