hbldh/bleak #243: Using bleak for HID emulation

IamBeardo, 2020-07-02:
#### question ####

Can Bleak be used to emulate a HID device (mouse)?

Specifically i want to advertise the emulator as a mouse, and initiate pairing from an iPhone. 

The software would then emulate mouse events (iPhone touch events, in this context) in order to control iPhone app(s)

Sorry if this is not the right forum for questions like this one. 

/Adam

hbldh, 2020-07-02:
Hi, no I do not think so. Bleak is a package that enables connections to Bluetooth Low Energy devices.

If you can prove me wrong and emulate a HID device with it, I will be most impressed and reward you in some fashion.,

dhalbert, 2020-07-03:
The bleak API doesn't provide the capabilities to act as a BLE peripheral (e.g., advertising), which is what a BLE HID device would need to do.

Some or all of the underlying platform libraries do have these capabilities, but this has not been exposed via bleak.

IamBeardo, 2020-07-03:
Thanks for the clarification

closed 2020-07-09 by hbldh