hbldh/bleak #246: Bluetooth interface list

aleksmarcelo, 2020-07-03:
Hi,

  would be nice to have methods to get the all bluetooth interfaces and  select one of them to be used.

   Thank you for the awesome library!

hbldh, 2020-07-09:
I am not sure of what you mean here by "all the bluetooth interfaces". Do you mean GATT profiles?

aleksmarcelo, 2020-07-09:
<Linux> 
I meant to make Bleak to get one more method to return all the bluetooth interfaces  (hci0, hci0...) without using other library. I think Bleak already internally get this list, right ? 
</Linux>

dlech, 2020-07-10:
I think Linux is the only OS that allows using more than one Bluetooth adapter at a time.

What is the actual problem that you are trying to solve by getting the individual adapters?

aleksmarcelo, 2020-07-12:
Bleak already allows you to select the interface, so you, without import other dependency, would be able to select a any Bluetooth interface you want. 

For example, my notebook's internal bluetooth adapter is only 4.2 compliant, so i allways use a 5.0 Ble dongle. But in BLEAK is hardcoded the use of the first bluetooth interface forcing me do disable the internal bluetooth in order to use the dongle. 

It would solve this sort of problem, don't you think ?

hbldh, 2020-07-16:
To use a user specified adapter/interface, use the `BleakClient(address, device="hci1")` initialization.

closed 2020-09-25 by hbldh