hbldh/bleak #247: Reading vendor specific advertising data

RulerKun, 2020-07-06:
How do I read the advertising data from scanned BLE devices? I need to filter them with the Vendor Specific data

hbldh, 2020-07-09:
Discovery methods return `BLEDevice` objects. The `details` attribute contains the advertisement data returned by the OS. Look at https://github.com/hbldh/bleak/blob/master/bleak/backends/device.py#L12-L24 for more details regarding your specific OS and google from that. 

Additionally, the `metadata` attribute is a dict containing a `manufacturer_data`  key where that data is stored. Is that what you are looking for?

hbldh, 2020-09-25:
In Windows you can do this: https://github.com/hbldh/bleak/issues/118#issuecomment-685432658

Will close this for now. Feel free to reopen if you have details on how to do it in other backends and I will integrate it into the documentation in #266.

closed 2020-09-25 by hbldh