hbldh/bleak #248: Unable to install BLEAK in HomeAssistant

mavrikkk, 2020-07-07:
* bleak version: last
* Python version: 3.7
* Operating System: raspbian
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

Try to use your library in my integration. Add it to requrements.
Installation failed with the following log:

Unable to install package bleak: ERROR: Command errored out with exit status 1: command: /usr/local/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-vvwj78xu/twisted/setup.py'"'"'; __file__='"'"'/tmp/pip-install-vvwj78xu/twisted/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-k9eheqfm cwd: /tmp/pip-install-vvwj78xu/twisted/ Complete output (953 lines): running bdist_wheel running build running build_py creating build creating build/lib.linux-armv7l-3.7 creating build/lib.linux-armv7l-3.7/twisted copying src/twisted/copyright.py -> build/lib.linux-armv7l-3.7/twisted copying src/twisted/__main__.py -> build/lib.linux-armv7l-3.7/twisted copying src/twisted/plugin.py -> build/lib.linux-armv7l-3.7/twisted copying src/twisted/__init__.py -> build/lib.linux-armv7l-3.7/twisted copying src/twisted/_version.py -> build/lib.linux-armv7l-3.7/twisted creating build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/mock_win32process.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_ftp.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/stdio_test_lastwrite.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_main.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_logfile.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/testutils.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/process_tester.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_pcp.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/process_cmdline.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/reflect_helper_VE.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_sslverify.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_internet.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_adbapi.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_sip.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/proto_helpers.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/process_stdinreader.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_strports.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/plugin_extra2.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/process_signal.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_tcp_internals.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_reflect.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/stdio_test_loseconn.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/stdio_test_consumer.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/iosim.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/process_echoer.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/plugin_extra1.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/myrebuilder2.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_tcp.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_sob.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_process.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_stdio.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_application.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_compat.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/stdio_test_writeseq.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/process_linger.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_amp.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_postfix.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_task.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_log.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_lockfile.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_defer.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/process_getargv.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/process_tty.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_memcache.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_policies.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_twisted.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_threadpool.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_tpfile.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_formmethod.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_ssl.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_usage.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_unix.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_protocols.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/stdio_test_halfclose.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/myrebuilder1.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/ssl_helpers.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_dirdbm.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_iosim.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_ftp_options.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/process_reader.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_cooperator.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_dict.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_ident.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/plugin_basic.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_iutils.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_roots.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_nooldstyle.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_rebuild.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/__init__.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_fdesc.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/process_getenv.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_defgen.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_randbytes.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_failure.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_loopback.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/process_twisted.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_text.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/process_fds.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_htb.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_modules.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_threadable.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/stdio_test_write.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/reflect_helper_ZDE.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_strerror.py -> build/lib.linux-armv7l-3.7/twisted/test copying src/twisted/test/test_news.py -> buil [... cut ...]  -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-vvwj78xu/twisted/setup.py'"'"'; __file__='"'"'/tmp/pip-install-vvwj78xu/twisted/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-f4oml1g4/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.7m/twisted

```

mavrikkk, 2020-07-07:
However, if i try install it manually (pip3 install bleak) it will be installed

hbldh, 2020-07-09:
I have never tried bleak with HomeAssistant. 

It seems to want to build twisted by itself instead of installing from wheel available on PyPI. That should not be necessary.

Are you satisifed with the solution to install it manually instead? Is that normal for HomeAssistant integrations?

mavrikkk, 2020-07-27:
Unfortunately, it isnt normal for HA to install dependencies manually.

mavrikkk, 2020-07-27:
> It seems to want to build twisted by itself instead of installing from wheel available on PyPI

i see it. but i cant influence it

hbldh, 2020-07-29:
I am afraid that this falls outside of what I have time to look at. I will close this issue since there is a workaround that is not too problematic. Thank you for reporting it though!

closed 2020-07-29 by hbldh

FortezzaJ, 2025-02-20:
The workaround is very problematic as it cannot be installed manually with home assistant OS

dlech, 2025-02-20:
This issue has been closed for 5 years, so might not be up to date. Could you please start a new discussion explaining the problem you are having in more detail?