hbldh/bleak #249: ProtocolError (Error: 0x0F) on reading GATT characteristics

Gongxter, 2020-07-09:
* bleak version: 
* Python version: 3.7
* Operating System: Windows 10

### Description

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.
I'm trying to control my Philips Hue lamp via BLE. On some characteristics which have the properties of (read,write-without-response,write) this Protocol Error (0x0F) is thrown, so that I can't read the values of this characteristic. What does that error mean and how do I get to read the actual values?

### What I Did

``` python
devices = await discover()
    for d in devices:
        if d.name == "Hue Lamp":
            client = BleakClient(d.address)
            await client.connect()
            if await client.is_connected():
                services = await  client.get_services()
                for service in services:
                    print("[Service] {0}: {1}".format(service.uuid, service.description))
                    for char in service.characteristics:
                        if "read" in char.properties:
                            try:
                                value = bytes(await client.read_gatt_char(char.uuid))
                            except Exception as e:
                                value = str(e).encode()
                        else:
                            value = None
                        print(
                            "\t[Characteristic] {0}: (Handle: {1}) ({2}) | Name: {3}, Value: {4} ".format(
                                char.uuid,
                                char.handle,
                                ",".join(char.properties),
                                char.description,
                                value,
                            )
                        )
                        for descriptor in char.descriptors:
                            value = await client.read_gatt_descriptor(descriptor.handle)
                            print(
                                "\t\t[Descriptor] {0}: (Handle: {1}) | Value: {2} ".format(
                                    descriptor.uuid, descriptor.handle, bytes(value)
                                )
                            )
            else:
                print("failed to connect").
```

``` python
[Service] 00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile
	[Characteristic] 00002a05-0000-1000-8000-00805f9b34fb: (Handle: 2) (indicate) | Name: , Value: None 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 4) | Value: b'\x02\x00' 
[Service] 00001800-0000-1000-8000-00805f9b34fb: Generic Access Profile
	[Characteristic] 00002a00-0000-1000-8000-00805f9b34fb: (Handle: 6) (read) | Name: , Value: b'Hue Lamp' 
	[Characteristic] 00002a01-0000-1000-8000-00805f9b34fb: (Handle: 8) (read) | Name: , Value: b'\x00\x00' 
[Service] 0000180a-0000-1000-8000-00805f9b34fb: Device Information
	[Characteristic] 00002a29-0000-1000-8000-00805f9b34fb: (Handle: 11) (read) | Name: , Value: b'Philips' 
	[Characteristic] 00002a24-0000-1000-8000-00805f9b34fb: (Handle: 13) (read) | Name: , Value: b'LCA001' 
	[Characteristic] 00002a28-0000-1000-8000-00805f9b34fb: (Handle: 15) (read) | Name: , Value: b'1.53.3_r27175' 
[Service] 0000fe0f-0000-1000-8000-00805f9b34fb: Vendor specific
	[Characteristic] 97fe6561-0001-4f62-86e9-b71ee2da3d22: (Handle: 18) (read) | Name: , Value: b'{\xc8\xac\x08\x01\x88\x17\x00' 
	[Characteristic] 97fe6561-0003-4f62-86e9-b71ee2da3d22: (Handle: 20) (read,write-without-response,write) | Name: , Value: b'Could not get GATT characteristics for 97fe6561-0003-4f62-86e9-b71ee2da3d22: ProtocolError (Error: 0x0F)' 
	[Characteristic] 97fe6561-0004-4f62-86e9-b71ee2da3d22: (Handle: 22) (write-without-response,write) | Name: , Value: None 
	[Characteristic] 97fe6561-0008-4f62-86e9-b71ee2da3d22: (Handle: 24) (write-without-response,write,indicate) | Name: , Value: None 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 26) | Value: b'\x00\x00' 
	[Characteristic] 97fe6561-2001-4f62-86e9-b71ee2da3d22: (Handle: 27) (read,write-without-response,write) | Name: , Value: b'\x0c' 
	[Characteristic] 97fe6561-2002-4f62-86e9-b71ee2da3d22: (Handle: 29) (write-without-response,write) | Name: , Value: None 
	[Characteristic] 97fe6561-2004-4f62-86e9-b71ee2da3d22: (Handle: 31) (write-without-response,write) | Name: , Value: None 
	[Characteristic] 97fe6561-a001-4f62-86e9-b71ee2da3d22: (Handle: 33) (write-without-response,write) | Name: , Value: None 
[Service] 932c32bd-0000-47a2-835a-a8d455b859dd: Unknown
	[Characteristic] 932c32bd-0001-47a2-835a-a8d455b859dd: (Handle: 36) (read) | Name: , Value: b'Could not get GATT characteristics for 932c32bd-0001-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F)' 
	[Characteristic] 932c32bd-0002-47a2-835a-a8d455b859dd: (Handle: 38) (read,write-without-response,write,notify) | Name: , Value: b'Could not get GATT characteristics for 932c32bd-0002-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F)' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 40) | Value: b'\x00\x00' 
	[Characteristic] 932c32bd-0003-47a2-835a-a8d455b859dd: (Handle: 41) (read,write-without-response,write,notify) | Name: , Value: b'Could not get GATT characteristics for 932c32bd-0003-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F)' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 43) | Value: b'\x00\x00' 
	[Characteristic] 932c32bd-0004-47a2-835a-a8d455b859dd: (Handle: 44) (read,write-without-response,write,notify) | Name: , Value: b'Could not get GATT characteristics for 932c32bd-0004-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F)' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 46) | Value: b'\x00\x00' 
	[Characteristic] 932c32bd-0005-47a2-835a-a8d455b859dd: (Handle: 47) (read,write-without-response,write,notify) | Name: , Value: b'Could not get GATT characteristics for 932c32bd-0005-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F)' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 49) | Value: b'\x00\x00' 
	[Characteristic] 932c32bd-0006-47a2-835a-a8d455b859dd: (Handle: 50) (write-without-response,write) | Name: , Value: None 
	[Characteristic] 932c32bd-0007-47a2-835a-a8d455b859dd: (Handle: 52) (read,write-without-response,write,notify) | Name: , Value: b'Could not get GATT characteristics for 932c32bd-0007-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F)' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 54) | Value: b'\x00\x00' 
	[Characteristic] 932c32bd-1003-47a2-835a-a8d455b859dd: (Handle: 55) (read,write-without-response,write) | Name: , Value: b'Could not get GATT characteristics for 932c32bd-1003-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F)' 
	[Characteristic] 932c32bd-1005-47a2-835a-a8d455b859dd: (Handle: 57) (read,write-without-response,write) | Name: , Value: b'Could not get GATT characteristics for 932c32bd-1005-47a2-835a-a8d455b859dd: ProtocolError (Error: 0x0F)' 
[Service] b8843add-0000-4aa1-8794-c3f462030bda: Unknown
	[Characteristic] b8843add-0001-4aa1-8794-c3f462030bda: (Handle: 60) (read) | Name: , Value: b'Could not get GATT characteristics for b8843add-0001-4aa1-8794-c3f462030bda: ProtocolError (Error: 0x0F)' 
	[Characteristic] b8843add-0002-4aa1-8794-c3f462030bda: (Handle: 62) (write-without-response,write,indicate) | Name: , Value: None 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 64) | Value: b'\x00\x00' 
	[Characteristic] b8843add-0003-4aa1-8794-c3f462030bda: (Handle: 65) (write-without-response,write) | Name: , Value: None 
	[Characteristic] b8843add-0004-4aa1-8794-c3f462030bda: (Handle: 67) (read) | Name: , Value: b'Could not get GATT characteristics for b8843add-0004-4aa1-8794-c3f462030bda: ProtocolError (Error: 0x0F)' 

```

Matt-Hicks-Bose, 2020-07-09:
In the Bluetooth Core Specification (available on the BT SIG website: https://www.bluetooth.com/specifications/bluetooth-core-specification/ ), Version 5.2, Vol 3, Part H, Section 3.4.1.1, Table 3.4 (pp. 1491) the Error Code 0x0F send in an `ATT_ERROR_RSP` is "Insufficient Encryption - The attribute requires encryption before it can be read or written."

I believe this can be accomplished by pairing and bonding the BLE device to the computer before connecting to it (see: https://stackoverflow.com/questions/54369175/how-to-encrypt-data-using-ble-connection-using-bluez5-50 and https://stackoverflow.com/questions/42862560/should-one-create-a-bond-with-a-bluetooth-le-device ).

Gongxter, 2020-07-10:
@Matt-Hicks-Bose @hbldh Thanks alot for your help. So I paired now my computer over windows with the Hue Lamp and now it seems to be working.
``` python
[Service] 00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile
	[Characteristic] 00002a05-0000-1000-8000-00805f9b34fb: (Handle: 2) (indicate) | Name: , Value: None 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 4) | Value: b'\x00\x00' 
[Service] 00001800-0000-1000-8000-00805f9b34fb: Generic Access Profile
	[Characteristic] 00002a00-0000-1000-8000-00805f9b34fb: (Handle: 6) (read) | Name: , Value: b'Hue Lamp' 
	[Characteristic] 00002a01-0000-1000-8000-00805f9b34fb: (Handle: 8) (read) | Name: , Value: b'\x00\x00' 
[Service] 0000180a-0000-1000-8000-00805f9b34fb: Device Information
	[Characteristic] 00002a29-0000-1000-8000-00805f9b34fb: (Handle: 11) (read) | Name: , Value: b'Philips' 
	[Characteristic] 00002a24-0000-1000-8000-00805f9b34fb: (Handle: 13) (read) | Name: , Value: b'LCA001' 
	[Characteristic] 00002a28-0000-1000-8000-00805f9b34fb: (Handle: 15) (read) | Name: , Value: b'1.53.3_r27175' 
[Service] 0000fe0f-0000-1000-8000-00805f9b34fb: Vendor specific
	[Characteristic] 97fe6561-0001-4f62-86e9-b71ee2da3d22: (Handle: 18) (read) | Name: , Value: b'{\xc8\xac\x08\x01\x88\x17\x00' 
	[Characteristic] 97fe6561-0003-4f62-86e9-b71ee2da3d22: (Handle: 20) (read,write-without-response,write) | Name: , Value: b'Hue bulb' 
	[Characteristic] 97fe6561-0004-4f62-86e9-b71ee2da3d22: (Handle: 22) (write-without-response,write) | Name: , Value: None 
	[Characteristic] 97fe6561-0008-4f62-86e9-b71ee2da3d22: (Handle: 24) (write-without-response,write,indicate) | Name: , Value: None 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 26) | Value: b'\x00\x00' 
	[Characteristic] 97fe6561-2001-4f62-86e9-b71ee2da3d22: (Handle: 27) (read,write-without-response,write) | Name: , Value: b'\n' 
	[Characteristic] 97fe6561-2002-4f62-86e9-b71ee2da3d22: (Handle: 29) (write-without-response,write) | Name: , Value: None 
	[Characteristic] 97fe6561-2004-4f62-86e9-b71ee2da3d22: (Handle: 31) (write-without-response,write) | Name: , Value: None 
	[Characteristic] 97fe6561-a001-4f62-86e9-b71ee2da3d22: (Handle: 33) (write-without-response,write) | Name: , Value: None 
[Service] 932c32bd-0000-47a2-835a-a8d455b859dd: Unknown
	[Characteristic] 932c32bd-0001-47a2-835a-a8d455b859dd: (Handle: 36) (read) | Name: , Value: b'\x00\x01\x03\x01\x02\x99\x00\x02\x02\xf4\x01\x03\x02\x01\x00' 
	[Characteristic] 932c32bd-0002-47a2-835a-a8d455b859dd: (Handle: 38) (read,write-without-response,write,notify) | Name: , Value: b'\x01' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 40) | Value: b'\x00\x00' 
	[Characteristic] 932c32bd-0003-47a2-835a-a8d455b859dd: (Handle: 41) (read,write-without-response,write,notify) | Name: , Value: b'\xfe' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 43) | Value: b'\x00\x00' 
	[Characteristic] 932c32bd-0004-47a2-835a-a8d455b859dd: (Handle: 44) (read,write-without-response,write,notify) | Name: , Value: b'\xf4\x01' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 46) | Value: b'\x00\x00' 
	[Characteristic] 932c32bd-0005-47a2-835a-a8d455b859dd: (Handle: 47) (read,write-without-response,write,notify) | Name: , Value: b'\xce\x86\xcei' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 49) | Value: b'\x00\x00' 
	[Characteristic] 932c32bd-0006-47a2-835a-a8d455b859dd: (Handle: 50) (write-without-response,write) | Name: , Value: None 
	[Characteristic] 932c32bd-0007-47a2-835a-a8d455b859dd: (Handle: 52) (read,write-without-response,write,notify) | Name: , Value: b'\x01\x01\x01\x02\x01\xfe\x03\x02\xf4\x01\x04\x04\xce\x86\xcei' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 54) | Value: b'\x00\x00' 
	[Characteristic] 932c32bd-1003-47a2-835a-a8d455b859dd: (Handle: 55) (read,write-without-response,write) | Name: , Value: b'\x04\x00' 
	[Characteristic] 932c32bd-1005-47a2-835a-a8d455b859dd: (Handle: 57) (read,write-without-response,write) | Name: , Value: b'\x01\x01\x01\x02\x01\xfe\x03\x02n\x01\x04\x04\xff\xff\xff\xff' 
[Service] b8843add-0000-4aa1-8794-c3f462030bda: Unknown
	[Characteristic] b8843add-0001-4aa1-8794-c3f462030bda: (Handle: 60) (read) | Name: , Value: b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\x00' 
	[Characteristic] b8843add-0002-4aa1-8794-c3f462030bda: (Handle: 62) (write-without-response,write,indicate) | Name: , Value: None 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 64) | Value: b'\x00\x00' 
	[Characteristic] b8843add-0003-4aa1-8794-c3f462030bda: (Handle: 65) (write-without-response,write) | Name: , Value: None 
	[Characteristic] b8843add-0004-4aa1-8794-c3f462030bda: (Handle: 67) (read) | Name: , Value: b'\x01' 

```

I don't know if that is a bug or not but the is_connected function now returns false while I would expect the result to be true. 
```python
if await client.is_connected():
```
But apart from that everything works fine and after I removed the branch got the results as seen above.

Thanks alot for your help.

Gongxter, 2020-07-10:
Okay, somehow I cannot reproduce on how I established the secured connection with the lamp. Pairing the device does not seems to help, since the connection with the BLE device (the Hue Lamp) is now always unsecured.
I used now microsoft's "[Bluetooth LE Explorer](https://github.com/microsoft/BluetoothLEExplorer)". 
So Pairing via Windows Bluetooth properties as well as pairing via the app results in the same result of an unsecured connection.
[image]

The working attempt above had a secured connection (got displayed in the Bluetooth LE Explorer app) and I don't know how.

Gongxter, 2020-07-16:
I hoped to find a solution on how to establish a secure connection, but sadly I couldn't. I've also tried to get this done with my raspberry pi with the same results. Even with pairing the device and setting the sec-level to medium or high in gatttool on my raspberry didn't help. I'm by no means an expert in this area but this problem has nothing todo with this Library so I will close this Issue.

closed 2020-07-16 by Gongxter

hbldh, 2020-07-20:
@Gongxter I will hopefully get a pairing solution in bleak working in version 0.8.0, but it still might not solve your problem. Just so you know.