hbldh/bleak #250: Bleak is not returning Generic Access Service (0x1800) and  looks like the handles are wrong  (Linux implementation)

aleksmarcelo, 2020-07-09:
* bleak version: 0.7.1

* Python version: 3.6.9

* Operating System: LINUX  ( 5.4.0-40-generic #44~18.04.1-Ubuntu SMP Wed Jun 24 23:13:08 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux)

* BlueZ version (`bluetoothctl -v`) in case of Linux:  5.54

### Description

When you simply list services and characteristics ( I used a Mi Band Fit) 

### What I Did

Bleak version and results
```python
import asyncio
from bleak import BleakClient

async def run(address, loop):
    client = BleakClient(address, loop=loop)

    x = await client.connect()

    for service in client.services:
        print(service.uuid)
        for c in service.characteristics:
            print(f"\t[Characteristic] {c.uuid}:  Handle: {hex(c.handle)} ")

if __name__ == "__main__":
    address = "DB:6E:07:2F:66:38"
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run(address, loop))
```
## Results Bleak:
```
00003802-0000-1000-8000-00805f9b34fb
        00004a02-0000-1000-8000-00805f9b34fb:  Handle: 0x74 
0000fee1-0000-1000-8000-00805f9b34fb
        0000fec1-0000-3512-2118-0009af100700:  Handle: 0x70 
        0000fed3-0000-1000-8000-00805f9b34fb:  Handle: 0x6e 
        0000fed2-0000-1000-8000-00805f9b34fb:  Handle: 0x6c 
        0000fed1-0000-1000-8000-00805f9b34fb:  Handle: 0x6a 
        0000fed0-0000-1000-8000-00805f9b34fb:  Handle: 0x68 
        0000fedf-0000-1000-8000-00805f9b34fb:  Handle: 0x66 
        0000fede-0000-1000-8000-00805f9b34fb:  Handle: 0x64 
        0000fedd-0000-1000-8000-00805f9b34fb:  Handle: 0x62 
        00000009-0000-3512-2118-0009af100700:  Handle: 0x5f 
0000fee0-0000-1000-8000-00805f9b34fb
        00000013-0000-3512-2118-0009af100700:  Handle: 0x5b 
        00000012-0000-3512-2118-0009af100700:  Handle: 0x58 
        00000011-0000-3512-2118-0009af100700:  Handle: 0x55 
        0000000f-0000-3512-2118-0009af100700:  Handle: 0x52 
        0000000e-0000-3512-2118-0009af100700:  Handle: 0x4f 
        00000020-0000-3512-2118-0009af100700:  Handle: 0x4c 
        00000010-0000-3512-2118-0009af100700:  Handle: 0x49 
        00000008-0000-3512-2118-0009af100700:  Handle: 0x46 
        00000007-0000-3512-2118-0009af100700:  Handle: 0x43 
        00000006-0000-3512-2118-0009af100700:  Handle: 0x40 
        00000005-0000-3512-2118-0009af100700:  Handle: 0x3d 
        00000004-0000-3512-2118-0009af100700:  Handle: 0x3a 
        00002a04-0000-1000-8000-00805f9b34fb:  Handle: 0x37 
        00000003-0000-3512-2118-0009af100700:  Handle: 0x34 
        00000002-0000-3512-2118-0009af100700:  Handle: 0x31 
        00000001-0000-3512-2118-0009af100700:  Handle: 0x2e 
        00002a2b-0000-1000-8000-00805f9b34fb:  Handle: 0x2b 
0000180d-0000-1000-8000-00805f9b34fb
        00002a39-0000-1000-8000-00805f9b34fb:  Handle: 0x28 
        00002a37-0000-1000-8000-00805f9b34fb:  Handle: 0x25 
00001802-0000-1000-8000-00805f9b34fb
[... 122 lines cut ...]
        00000007-0000-3512-2118-0009af100700  handle:0x44 
        00000008-0000-3512-2118-0009af100700  handle:0x47 
        00000010-0000-3512-2118-0009af100700  handle:0x4a 
        00000020-0000-3512-2118-0009af100700  handle:0x4d 
        0000000e-0000-3512-2118-0009af100700  handle:0x50 
        0000000f-0000-3512-2118-0009af100700  handle:0x53 
        00000011-0000-3512-2118-0009af100700  handle:0x56 
        00000012-0000-3512-2118-0009af100700  handle:0x59 
        00000013-0000-3512-2118-0009af100700  handle:0x5c 
0000fee1-0000-1000-8000-00805f9b34fb 
        00000009-0000-3512-2118-0009af100700  handle:0x60 
        0000fedd-0000-1000-8000-00805f9b34fb  handle:0x63 
        0000fede-0000-1000-8000-00805f9b34fb  handle:0x65 
        0000fedf-0000-1000-8000-00805f9b34fb  handle:0x67 
        0000fed0-0000-1000-8000-00805f9b34fb  handle:0x69 
        0000fed1-0000-1000-8000-00805f9b34fb  handle:0x6b 
        0000fed2-0000-1000-8000-00805f9b34fb  handle:0x6d 
        0000fed3-0000-1000-8000-00805f9b34fb  handle:0x6f 
        0000fec1-0000-3512-2118-0009af100700  handle:0x71 
00003802-0000-1000-8000-00805f9b34fb 
        00004a02-0000-1000-8000-00805f9b34fb  handle:0x75 
```

You can observe that the service 0x1800  is not returned by bleak and the handle id is not the right one pointing to the characteristic.
Anyway,  correcting some issues Bleak will be the most one!

hbldh, 2020-07-16:
The BlueZ DBus API does things a bit differently from e.g. `gatttool` and previous BlueZ API solutions. The Battery Service (0x180F) is instead served under the Battery DBus interface, making that service not available in `get_services` calls. That service has, I believe, been relocated to the Device interface and is available on the device object instead. See here: https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/device-api.txt

The device you have connected to is available in object you get when doing discovery. Check the `details` attribute of the `BLEDevice` object.

Bleak tries its hardest to be identical over all OS:es, but some things are not possible to do, and I think this is one of them. The BlueZ DBus API is **not** comparable to `gatttool`. Are the handles you use giving you the wrong data? Verify that first. If they differ it might be due to BlueZ readressing this internally, I cannot be certain. Bleak merely uses what handles that are presented by the BlueZ DBus API.

hbldh, 2020-09-25:
There has been some changes regarding handles in version 0.8.0. It might solve this, if it even was an issue to start with. I get correct results using the handles that Bleak provides, at least for my devices. They are the handles provided by the BlueZ DBus API and I cannot use anything else than them.

Will close this issue. Feel free to reopen if new data is dicovered.

closed 2020-09-25 by hbldh