hbldh/bleak #252: cannot connect under MacOS (10.15.4)

danaronson, 2020-07-13:
* bleak version: 0.7.1
* Python version: 3.8.2
* Operating System: MacOS (10.15.4)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

Cannot connect to a device (this works fine on my pi).  I installed blueutil to see the osx can see the device.

### What I Did

```
bash-3.2# blueutil --info C6:7F:B8:C2:23:21
address: c6-7f-b8-c2-23-21, not connected, not favourite, not paired, name: "AirGo_2123c2b87fc6", recent access date: -
bash-3.2# python airgo.py --address C6:7F:B8:C2:23:21 --log-level INFO
WARNING:bleak.backends.corebluetooth.CentralManagerDelegate:CentralManagerDelegate is not compliant
Traceback (most recent call last):
  File "airgo.py", line 253, in <module>
    loop.run_until_complete(run(loop))
  File "/Users/dan/anaconda3/envs/py3/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "airgo.py", line 235, in run
    async with BleakClient(address, loop=loop) as client:
  File "/Users/dan/anaconda3/envs/py3/lib/python3.8/site-packages/bleak/backends/client.py", line 49, in __aenter__
    await self.connect()
  File "/Users/dan/anaconda3/envs/py3/lib/python3.8/site-packages/bleak/backends/corebluetooth/client.py", line 77, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address C6:7F:B8:C2:23:21 was not found
```

dlech, 2020-07-13:
Please see https://bleak.readthedocs.io/en/latest/backends/macos.html

macOS does not allow access to Bluetooth addresses.

danaronson, 2020-07-13:
D'oh.  Thank you.  Still having issues, but not this one.

closed 2020-07-13 by danaronson