hbldh/bleak #256: not able to connect because ServicesResolved never gets set to true

danaronson, 2020-07-15:
* bleak version: 0.7.1
* Python version: 3.7.3
* Operating System: raspbian (raspberrypi 4.19.118-v7l+)
* BlueZ version (`bluetoothctl -v`) in case of Linux:  5.54

### Description

I cannot connect to a device, it gets a services discover error, looking at the bleak code, ServicesResolved is never getting set to true, however i can see this device (via hcitool) and I can connect to it (via either bluetoothctl or gatttool).

### What I Did

```
Traceback (most recent call last):
  File "airgo.py", line 250, in <module>
    loop.run_until_complete(run(loop))
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "airgo.py", line 232, in run
    async with BleakClient(address, loop=loop) as client:
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/client.py", line 49, in __aenter__
    await self.connect()
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 175, in connect
    await self.get_services()
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 317, in get_services
    raise BleakError("Services discovery error")
bleak.exc.BleakError: Services discovery error
```

hbldh, 2020-07-19:
Does your device have a lot of services and characteristics? Does this happen to all devices you connect to with that RPi or just one?

It might be timing out in bleak while waiting for services to be discovered. Could you, in you installed copy of bleak, change change row 308 of `bleak/backends/bluezdbus/client.py` from

```python
while total_slept_sec < 5.0:
```

to

```python
while total_slept_sec < 30.0:
```

This way we might establish if it is due to bleak waiting to short a time. In any case this is a bug, or in the very least a negligence to be able to set that value, and needs modification.

danaronson, 2020-07-19:
My suspicion right now is that it might be something going on in bluez.
Our firmware (which uses a UART) closes the connection after 3
seconds unless it gets a known key on the UART (so I don't think increasing
this to 30 seconds should do any better in discovering the services).   I
am awaiting our firmware guys to remove this to help me debug.  I'm going
to put this on hold until I get the changes so I'm not spinning anyone's
wheels, however anyone has suggestions on using hcidump or some other tool
to try to figure out how my two RPs (one a 4BRev1.1 and one a 4BRev1.2) are
differing.

hbldh, 2020-07-20:
Ok. I will wait with making any changes for this then, until you have looked further into this.

Bleak is quite slow getting services, characteristics and descriptors. It is not good, but it is something that I right now have no idea on how to improve.

danaronson, 2020-07-21:
so the good news is that i am able to discover services and characteristics using gatttool prior to our firmware closing the connection (which it does in 3 seconds).  However even when moving to 30 seconds timeout and 0.01 seconds for sleep I can't do it, which leads me to believe that bluez is getting them when using bleak.  As I said, probably a version of the firmware that doesn't close the connection will work, but ideally we don't want to put that out in the field.  Any suggestions on how to debug?

danaronson, 2020-07-22:
Good news and bad news.  I've identified the problem (good), don't know how to rectify it.

On the RPi where it isn't working, doing an hcidump is showing me that it is trying to enable encryption and then getting an error getting the key and closing the connection.

```
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
> HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
> HCI Event: Command Complete (0x0e) plen 14
    LE Read Remote Used Features (0x08|0x0016) ncmd 1
    . . . . . . . . . . .
> HCI Event: Command Status (0x0f) plen 4
    LE Start Encryption (0x08|0x0019) status 0x00 ncmd 1
> HCI Event: Encrypt Change (0x08) plen 4
    status 0x06 handle 64 encrypt 0x00
    Error: PIN or Key Missing
> HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
```

When I run this on the good RPi I only see:
```
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
> HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
> HCI Event: Command Complete (0x0e) plen 14
    LE Read Remote Used Features (0x08|0x0016) ncmd 1
    . . . . . . . . . . .
> HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x22
    Reason: LMP Response Timeout
```

danaronson, 2020-07-22:
I downgrade my raspbian kernel to 4.19.97-v7l (it was up to 118 on the one that didn't work) and now it works.

hbldh, 2020-07-29:
You only downgraded the kernel? Not BlueZ itself?

Do you regard this issue as closed? Are you satisifed with the solution? I am not sure that I can do anything about it anyway...

danaronson, 2020-07-29:
yup, still 5.50.  Still working through it.

avassdal, 2020-09-13:
I'm still getting this error as well

hbldh, 2020-09-25:
Will close this due that it seems to be outside of Bleak.

closed 2020-09-25 by hbldh

Nosugrof, 2025-02-18:
I just ran into this issue on Ubuntu 22.04. I'm using Pixel Buds Pro 2 ear buds and they work fine when I connect in bluetoothctl. If I allow the auto-detect to make the connection it works for ~3 seconds and then drops. If I then open bluetoothctl I can see the device shows connected. If I run a connect command I get "[CHG] Device 5C:33:7B:6C:E9:49 ServicesResolved: yes" and then it works without dropping out. If this issue doesn't belong here can you tell me where this should be reported?

Thanks

Nosugrof, 2025-02-18:
I just ran into this issue on Ubuntu 22.04. I'm using Pixel Buds Pro 2 ear buds and they work fine when I connect in bluetoothctl. If I allow the auto-detect to make the connection it works for ~3 seconds and then drops. If I then open bluetoothctl I can see the device shows connected. If I run a connect command I get "[CHG] Device 5C:33:7B:6C:E9:49 ServicesResolved: yes" and then it works without dropping out. If this issue doesn't belong here can you tell me where this should be reported?