hbldh/bleak #257: DotNet error during connection after Windows 10 May 2020 Update

SamerAbdelmoeti, 2020-07-16:
* bleak version: 0.7.1
* Python version: 3.7.7
* Operating System: Windows 10 Version 2004 (May 2020 Update)

### Description

- After Windows 10 May Update (version 2004), connection with ble peripheral returns a _BleakDotNetTaskError_ during the 'Get Services' step. Therefore, (connecting, get services, service explorer) don't work, but (discover) works fine.
- The returned error is: 'bleak.exc.BleakDotNetTaskError: System.Exception: Catastrophic failure (Exception from HRESULT: 0x8000FFFF (E_UNEXPECTED))'.
- Confirmed that the root cause occurs due to Windows 10 May update by doing a clean install of Windows 10 version 1909 (November 2019) where everything works perfectly.
- Peripheral is Arduino nano 33 BLE device with a Nordic nRF52840 microprocessor. 
- On the peripheral side, the connection is established even though the central fails to connect.

### What I Did
Run any example code that attempts to connect with the peripheral (all except discover.py).
```
File "myenv\lib\site-packages\bleak\backends\client.py", line 49, in __aenter__
    await self.connect()
  File "myenv\lib\site-packages\bleak\backends\dotnet\client.py", line 150, in connect
    services = await self.get_services()
  File "myenv\lib\site-packages\bleak\backends\dotnet\client.py", line 237, in get_services
    loop=self.loop,
  File "myenv\lib\site-packages\bleak\backends\dotnet\utils.py", line 73, in wrap_IAsyncOperation
    raise BleakDotNetTaskError(op.ErrorCode.ToString())
bleak.exc.BleakDotNetTaskError: System.Exception: Catastrophic failure (Exception from HRESULT: 0x8000FFFF (E_UNEXPECTED))
```

dlech, 2020-07-16:
Did you report the problem to Microsoft?

hbldh, 2020-07-20:
@SamerAbdelmoeti This is similar to #222. If you are able to, try using a different bluetooth adapter. There is nothing I can do in bleak to alleviate this I am afraid.

dlech, 2020-09-14:
FWIW, I have seen the `Catastrophic failure` error too on Windows 10 2004. I haven't had time to troubleshoot yet.

BriFuture, 2021-03-20:
It seems that Bluetooth Device/ BLE  Version is not match.
For example, the bluetooth peripherals is Bluetooth 4.0，as windows showed :
[image]

If I connect another BLE device (BLE4.0), bleak works fine, but if the device is BLE5.0, `Catastrophic failure` will occur.

hbldh, 2021-03-20:
@BriFuture Could you please follow [this guide](https://support.microsoft.com/en-us/windows/what-bluetooth-version-is-on-my-pc-f5d4cff7-c00d-337b-a642-d2d23b082793) and see if your computer's Bluetooth hardware supports Bluetooth 5.0?

BriFuture, 2021-03-20:
From the guide, the bluetooth firmware of LMP is 6.x, so the hardware is bluetooth 4.0, 

The windows bluetooth device manager failed to connect my BLE5.0 device, so the problem may be caused by the driver.

dlech, 2021-11-27:
We have dropped support for the dotnet backend. Please try with Bleak v0.13 or newer which uses the winrt backend.

closed 2021-11-27 by dlech