hbldh/bleak #259: AttributeError raised when calling client.is_connected() after BLE device is no longer communicating

Matt-Hicks-Bose, 2020-07-24:
* bleak version: 0.7.1
* Python version: 3.7.3
* Operating System: Raspberry OS (Debian Linux)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

I am seeing an `AttributeError` exception thrown under Linux when attempting to call `client.is_connected()` on a client who's BLE device is no longer communicating (turned off or out of range). I am not seeing this exception on Windows.

### What I Did

Minimal Example:
```python
import asyncio
import logging

from bleak import BleakClient

logging.basicConfig(level=logging.DEBUG)

async def run(address):
    log = logging.getLogger(__name__)

    async with BleakClient(address) as client:
        x = await client.is_connected()
        log.info('Connected: {0}'.format(x))

        while True:
            print(await client.is_connected())
            await asyncio.sleep(0.1)

if __name__ == '__main__':
    address = # REDACTED
    asyncio.run(run(address))
```

Once I started to see "True" repeated on the command line, I turned off the BLE device that was connected, and that results in the following log with an exception raised about 2 seconds after the BLE device has been powered off (MAC address has been redacted):
```python
INFO:__main__:Connected: True
True
True
...
True
True
DEBUG:bleak.backends.bluezdbus.client:DBUS: path: /org/bluez/hci0/dev_REDACTED, domain: org.bluez.Device1, body: {'ServicesResolved': False, 'Connected': False}
DEBUG:bleak.backends.bluezdbus.client:Device REDACTED disconnected.
DEBUG:bleak.backends.bluezdbus.client:Removing rule PropChanged, ID: 1
DEBUG:bleak.backends.bluezdbus.client:Disconnecting from BLE device...
Traceback (most recent call last):
  File "disconnect_test.py", line 21, in <module>
    asyncio.run(run(address))
  File "/usr/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "disconnect_test.py", line 16, in run
    print(await client.is_connected())
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 282, in is_connected
    return await self._bus.callRemote(
AttributeError: 'NoneType' object has no attribute 'callRemote'
```

The same exception is raised if the BLE device is taken out of range or put into an RF shielded box as well. I have on occasion seen this successfully print 'False' once or twice before raising the exception as well when polling `client.is_connected()` at a faster rate, but it is inconsistent.

I have also seen this happen by calling `client.disconnect()` in my application after the BLE device is no longer communicating, but it is not consistent and I am unable to reproduce this with a minimal example at the moment, it looks like the root cause is likely the same as above though.
```python
Traceback (most recent call last):
...
  File "/home/pi/repos/REDACTED/REDACTED/device.py", line 521, in disconnect
    await self.client.disconnect()
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 265, in disconnect
    is_disconnected = not await self.is_connected()
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 282, in is_connected
    return await self._bus.callRemote(
AttributeError: 'NoneType' object has no attribute 'callRemote'
```

hbldh, 2020-07-29:
This should be possible to handle better. Will try to get this done in version 0.8.0.

Matt-Hicks-Bose, 2020-08-14:
@hbldh I got another one today on the BlueZ backend in the same case, calling `is_connected()` in my application code:
```python
<snip from our application traceback>
    connected = await self.client.is_connected()
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 290, in is_connected
    ).asFuture(self.loop)
twisted.internet.error.ConnectionDone: Connection was closed cleanly.
```

I can occasionally cause this one by removing power from the connected BLE device manually. I am now protecting against both the ConnectionDone error and the AttributeError when calling `is_connected()`, and it seems to be working now. Seems related, but let me know if you want a new issue for this one that is separate.

commit 2456485c (2020-08-17): Improved is_connected in BlueZ backend to fix #259
closed 2020-09-14 by hbldh