hbldh/bleak #261: Error when trying to update a Descriptor with Start_notify or write_gatt_descriptor

McPatPyhton, 2020-08-04:
* bleak version: 0.7.1
* Python version: 3.8.5
* Operating System: Windows 10

### Description
Hello,

Not sure where the issue is, between me, the device and bleak but i am attempting to get notification from a ble thermoter on a Windows 10 pc with bluetooth.

I took a look into the android app code that communicate with this BLE device and i found that, when setting a notification, they directly updated this Descriptor with his UUID. No other UIID are referenced in the code.
"
	this.B.setCharacteristicNotification(bluetoothGattCharacteristic, A);

        BluetoothGattDescriptor descriptor = bluetoothGattCharacteristic.getDescriptor(UUID);
        if (A) {
            descriptor.setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
        } else {
            descriptor.setValue(BluetoothGattDescriptor.DISABLE_NOTIFICATION_VALUE);
        }
        this.B.writeDescriptor(descriptor);
"
This Descriptor UUID is attached to a Characteristic of the type "Indicate". If i use start_notify on this Characteristic  i get the exception "System.UnauthorizedAccessException: Accès refusé (Exception de HRESULT : 0x80070005 (E_ACCESSDENIED))" 

Then if i try to directly update the corresponding Descriptor with "write_gatt_descriptor", i get the "error object does not implement IAsyncOperation`1"

Note that i am to use the function start_notify on the Characteristic of the type "Notify" without error but i do not receive any notification. 

So, is there a way to update the Descriptor of the Characteristic "Indicate" with Bleak with other functions than write_gatt_descriptor? 

Regards,
Patrick

hbldh, 2020-08-09:
You might need to pair with the device, something that will be added to Bleak in 0.8.0 before being able to use `start_notify` on it. The Windows UWP BLE API is different from the Android one and I cannot guarantee the same functionality just because it connects to the same device.

I am not sure here, but try to pair with it using Windows Bluetooth solutions before connecting in Bleak and see if that changes anything.

hbldh, 2020-09-02:
This might be solved by #269. Try using the `pair` method when the 0.8.0 release hits PyPI.

hbldh, 2020-09-14:
Thi could be closed by #279 and verion 0.8.0. Reopen if not.

closed 2020-09-14 by hbldh