hbldh/bleak #263: Support for long reads/writes?

chris1seto, 2020-08-07:
Just wondering if support for long reads/writes is planned? Trying to write_gatt_char with a bytearray larger than the mtu results in a:

`
Connected: True
Traceback (most recent call last):
  File "./ble2.py", line 30, in <module>
    loop.run_until_complete(run(address, loop, True))
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "./ble2.py", line 21, in run
    await client.write_gatt_char('0000abf1-0000-1000-8000-00805f9b34fb', bytearray(text, 'utf-8'))
  File "/home/chris/.local/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 540, in write_gatt_char
    await self._bus.callRemote(
txdbus.error.RemoteError: org.bluez.Error.Failed: Failed to initiate write
`

dlech, 2020-08-07:
What is your version info?
* bleak version:
* Python version:
* Operating System:
* BlueZ version (`bluetoothctl -v`) in case of Linux:

chris1seto, 2020-08-07:
* Bleak 0.7.1 from pip3
* python 3.8.2
* Ubuntu 20.04.1 LTS
* BlueZ 5.53

Thanks!

hbldh, 2020-08-09:
`Failed to initiate write`? Neer seen that one before...

There is a branch called `feature/linux-mtu` in which mtu can be negiotiated with the periheral. Try that one out and see if you can get a "longer" write to work.

chris1seto, 2020-08-09:
@hbldh note that a "long write" is a specific btle transaction. During a long write, a series of "write preps" are performed followed by a a write excecute. This is regardless of mtu. Even with an mtu of 20 bytes, an entire gigabyte could be transferred this way.

hbldh, 2020-08-10:
If it is a specific transaction, then BlueZ needs to support it before Bleak can do it. Right now there is [only a write method](https://kernel.googlesource.com/pub/scm/bluetooth/bluez/+/refs/tags/5.53/doc/gatt-api.txt#89) and it does not seem to have any capacity to do what you request. You will have to experiment yourself with the methods that are available or adhere to the apparent limitations (not larger than mtu). Have you confirmed that writing below mtu limit works fine?

chris1seto, 2020-08-10:
@hbldh You can actually see that BlueZ supports this, given the "offset" argument.

A good high level oveview of long reads/writes can be found here: https://www.oreilly.com/library/view/getting-started-with/9781491900550/ch04.html under "Reading Characteristics and Descriptors"

FWIW: pyGATT will actually do a long write implicitly if if you give it a buffer ST sizeof(buffer) > mtu. It won't do a long read though. I suspect BlueZ will do a long write under the hood , but a long read needs to know how much data is being awaited, which I don't think pyGATT wrote support for.

I can't find where BlueZ talks about its support of long read/writes (the generic name of "long read" doesn't help...), but I think it is certainly there. See this project as well: https://github.com/gbrault/gattclient  . Note the argument "write-long-value"

hbldh, 2020-08-11:
@chris1seto The `offset` argument is implemented in the BlueZ backend on the `feature/linux-mtu` branch, so you are free to try it out there. I have no device to test this on, so I won't implement it myself I am afraid. I will review and merge pull requests on this topic if you are interested to implement and test it yourself though; you have already done most of the pre-implementation detective work.

However, one could just do a async for loop and calling the read/write method using the `offset` keyword and have this done without changing any code in Bleak.

chris1seto, 2020-08-11:
Thanks @hbldh I'll look into PR'ing this upstream to Bleak if it turns out that I end up needing to really use long reads and writes on my project.

robert-alfaro, 2021-12-18:
I've tried the linux-mtu branch... trying to get a long read working on one of my characteristics.  Setting MTU kwarg doesnt cause a problem but also doesnt seem to do anything.  Setting offset to anything > 0 causes error `org.bluez.Error.InvalidArguments: Invalid arguments in method call`.  This is true in the old branch as well as that same branch rebased with develop. After rebasing I got an error about Variant type so I made the value assigned to `body["offset"]` equal to `Variant('q', int(kwargs.get('offset')))` which gets rid of the previously mentioned InvalidArguments error, however, I get incomplete/corrupted reads from then on.

dlech, 2022-07-26:
There are no OS APIs for these low-level requests. If a write (with response) doesn't fit in a single packet, the OS Bluetooth stack should split it up into a long write automatically.

closed 2022-07-26 by dlech