hbldh/bleak #264: error when I combine bleak with gui

Szymon-Gesicki, 2020-08-07:
* bleak version: 0.7.1
* Python version: 3.7.7
* Operating System: MacOS Catalina 10.15.4
### Description

Hi, I have a problem. I saw another issue about tkkinter. I'm trying to connect the bleak library with the graphics library. On windows everything works great, unfortunately again I have a problem with macOS. If it's not the fault of the Bleak, I'm sorry.

### What I Did

I send a code that scans and then connects and displays the notified data.

Very often ( 75% ) times when I connect I get a error:

```
File "main.py", line 251, in <module>
    loop.run_until_complete(run(device, loop))
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/asyncqt/__init__.py", line 309, in run_until_complete
    return future.result()
  File "main.py", line 237, in run
    async with BleakClient(address, loop=loop) as client:
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/bleak/backends/client.py", line 49, in __aenter__
    await self.connect()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/bleak/backends/corebluetooth/client.py", line 83, in connect
    manager = self._device_info.manager().delegate()
AttributeError: 'NoneType' object has no attribute 'delegate'

```

```
import sys
import asyncio
import os

from PyQt5.QtWidgets import QApplication, QProgressBar, QFileDialog
from PyQt5 import QtCore, QtWidgets
from PyQt5.QtWidgets import QMainWindow, QLabel, QGridLayout, QWidget, QPushButton, QListWidget, QListWidgetItem
from PyQt5.QtCore import QSize 
from asyncqt import QEventLoop, QThreadExecutor
from PyQt5.QtWidgets import QDialog

import bleak
from bleak import discover, BleakClient

class MainWindow(QDialog):
    def __init__(self):
        super(MainWindow, self).__init__(None)

        self.setMinimumSize(QSize(500, 500))    
        self.setWindowTitle("Bleak GUI")         
        
        self.lDeviceCapt = QLabel("Device:", self)
        self.lDeviceCapt.resize(50, 32)
        self.lDeviceCapt.move(50, 50)
        
        self.lDeviceInfo = QLabel("[xxx]", self)
        self.lDeviceInfo.resize(200, 32)
        self.lDeviceInfo.move(200, 50)
        
        self.lCurrMessage = QLabel("", self)
        self.lCurrMessage.resize(200, 32)
        self.lCurrMessage.move(200, 100)
        
        self.lMessages = QListWidget(self)
        self.lMessages.resize(200, 300)
        self.lMessages.move(200,150)
        self.started = False
        
    def notify(self, msg):
        self.lMessages.addItem(str(msg))

    def did_connect(self, address):
        self.lDeviceInfo.setText(str(address))
    
app = QApplication(sys.argv)
loop = QEventLoop(app)
asyncio.set_event_loop(loop)

app = QtWidgets.QApplication(sys.argv)
mainWin = MainWindow()
mainWin.show()

devices_list = []

async def scan():
    devices = await discover()
    for d in devices:
        print(d)
        devices_list.append(d)

loop = asyncio.get_event_loop()
loop.run_until_complete(scan())
	
device = devices_list[0].address

def callback(sender, data):
    print(f"{sender}: {data}")
    mainWin.notify(data)

async def run(address, loop):

     async with BleakClient(address, loop=loop) as client:
        
        mainWin.did_connect(address)

        await client.start_notify(YOUR_UUID, callback)

        while True:
            await asyncio.sleep(5.0)
            if not await client.is_connected():
                print("Disconnect")
                break

with loop:
    loop.run_until_complete(run(device, run))

```

hbldh, 2020-08-09:
Can you run the Bleak code without the Qt interface?

I am afraid I have not got the time to make Bleak work alongside GUI frameworks at the moment. If you can find a way I am open to PRs.

Carglglz, 2020-08-10:
@Szymon-Gesicki 
`AttributeError: 'NoneType' object has no attribute 'delegate'`

This is due to the **client not connecting** to the device so `self._device_info.manager()` returns `None`  instead of the `manager`
and therefore `None` has no `delegate` attribute. This is why it sometimes works (the device is connected) and sometimes it doesn't work (the device didn't connect).

Use this instead of the context manager, where the connection timeout can be increased:
```
async def run(address, loop):
    client = BleakClient(address, loop=loop)
    try:
        await client.connect(timeout=10) # This will increase the odds to discover and connect to the device
        model_number = await client.read_gatt_char(MODEL_NBR_UUID)
       [....]
````

@hbldh 

> make Bleak work alongside GUI frameworks at the moment

I've actually been testing PyQt5 alongside `bleak_sigspec` development and everything works great. I will release the concept App/Tool when the `bleak_sigspec` PR is done (It needs the new set of features)

dlech, 2020-08-10:
Is there a reason to use Bleak instead of QBluetooth if using PyQt already?

Moncax, 2020-08-11:
> @Szymon-Gesicki
> `AttributeError: 'NoneType' object has no attribute 'delegate'`
> 
> This is due to the **client not connecting** to the device so `self._device_info.manager()` returns `None` instead of the `manager`
> ,therefore `None` has no `delegate` attribute. This is why sometimes it works (the device is connected) and sometimes it doesn't work (the device didn't connect).
> 
> Use this instead of the context manager, where the connection timeout can be increased:
> 
> ```
> async def run(address, loop):
>     client = BleakClient(address, loop=loop)
>     try:
>         await client.connect(timeout=10) # This will increase the odds to discover and connect to the device
>         model_number = await client.read_gatt_char(MODEL_NBR_UUID)
>        [....]
> ```
> 
> @hbldh
> 
> > make Bleak work alongside GUI frameworks at the moment
> 
> I've actually been testing PyQt5 alongside `bleak_sigspec` development and everything works great. I will release the concept App/Tool when the `bleak_sigspec` PR is done (It needs the new set of features)

I hope I can see your code soon. It can be a good example that I really thirst.

Moncax, 2020-08-11:
> @Szymon-Gesicki
> `AttributeError: 'NoneType' object has no attribute 'delegate'`
>
> This is due to the **client not connecting** to the device so `self._device_info.manager()` returns `None` instead of [...]

I hope I can see your code soon. It can be a good example that I really thirst for.

Carglglz, 2020-08-11:
@dlech 
I know QBluetooth exists but it is not documented in PyQt docs, (it is in Qt with examples in C++). Also I think it is at the same level of PyObjC Core Bluetooth, so a backend will be needed first (like `bluezdbus`, `dotnet`, `corebluetooth`) to manage scanning/ connection/ services/ read/write gatt operations as easy as bleak.

@Moncax 
I've just uploaded some pics: 
[bleappdemo.png](https://github.com/Carglglz/bleak_sigspec/blob/master/docs/img/bleappdemo.png)

[bleappdemo_hover.png](https://github.com/Carglglz/bleak_sigspec/blob/master/docs/img/bleappdemo_hover.png)

[bleappdemo_logging.png](https://github.com/Carglglz/bleak_sigspec/blob/master/docs/img/bleappdemo_logging.png)

The basic idea It's a system tray/ menu bar app which works (or at least I tried) like nRF Connect. 
The icon is set according to device appearance characteristic ( in Device Information Service) and menu is created following the services and characteristics that the device exposes.
Also everything is logged to sys.stdout so any action/error can be debugged. 👍

dlech, 2020-08-11:
QBluetooth is basically the same as Bleak - it provides a common abstraction for cross-platform backends. You do not have to create your own backend. I've used it successfully with PyQt before. I didn't think it was too difficult to translate the C++ examples to Python. Using QBluetooth seemed easier than trying to integrate asyncio into the Qt run loop.

Carglglz, 2020-08-11:
Sorry I didn't mean backend, I actually meant create bindings to QBluetooth like bleak have to the different backends. 
From this example : [qtbluetooth-le-overview](https://doc.qt.io/qt-5/qtbluetooth-le-overview.html) it looks like one should implement `client.py`, `service.py`, `characteristic.py`, `scanner.py`, `device.py` to work like bleak? Maybe I'm wrong, but since there is no PyQt5 QBluetooth docs I didn't even try...

> Using QBluetooth seemed easier than trying to integrate asyncio into the Qt run loop

I use QThreads to run the asyncio event loop, the data is passed to the Qt run loop by signals.
This way the GUI and bleak run independently and it seems to work quite good.

hbldh, 2020-09-25:
This project might be of use: https://pypi.org/project/qasync/ 

Hopefully I have time to add a Qt example with Bleak when working on #266. In the meantime, I will close this issue.

closed 2020-09-25 by hbldh

ivanjimenez, 2022-07-07:
> QBluetooth is basically the same as Bleak - it provides a common abstraction for cross-platform backends. You do not have to create your own backend. I've used it successfully with PyQt before. I [...]

I'm trying to integrate bleak, pyqt5, asyncio, qasync. But I don't think that qasync will be maintened in the future with support for Pyqt6 or above. PyQTBlueTooth is poor documented. I have scanned and connected a IMU sensor but I don't know if you could publish some code using QtBluetooth for Python with examples with write_gatt_char, read_gatt_char, etc.

I apreciate your help.
Thanks!!