hbldh/bleak #265: Notification.py example issue

khawajamechatronics, 2020-08-09:
* bleak version: 0.7.1
* Python version: Python3.5
* Operating System: Raspbian Stretch
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

I am trying to connect bleak to a device for data communication using characteristics. UUID 6e400003-b5a3-f393-e0a9-e50e24dcca9e

I am using code enable Notification.py example.

After putting UUID and Device address, I run the code, it shows connection successful. I also able to send some data from the device. But it get disconnects after few seconds automatically.

[image]

[image]

khawajamechatronics, 2020-08-09:
I am also testing same code with Windows, here is output on windows 
Case1: Normally it doesnt detect the device.

[image]
Case 2: When I soft reset my device and instantly run the code then it doesn connects and receive some data but then disconnects automatically.

[image]

hbldh, 2020-08-09:
The example file is written in a way that it [disconnects after 5 seconds](https://github.com/hbldh/bleak/blob/master/examples/enable_notifications.py#L45), so that it disconnects is entirely by purpose. Nohting is wrong here, it is just the way the program, python and `asyncio` works. Go through e.g. [this primer on `asyncio`](https://realpython.com/async-io-python/) and see if you can understand why.

closed 2020-08-09 by hbldh

khawajamechatronics, 2020-08-09:
Thanks for prompt response, I will study async IO in details. And also Thank you for awesome Work!

khawajamechatronics, 2020-08-28:
Hi, Although library is very useful but can we get example running without Asyncio?

hbldh, 2020-08-29:
Hi, I am sorry to say that no, you can't. The `asyncio` is the driving force of the package and cannot be avoided. BLE is asynchronous in its very definition and works best in that framing.

If you want a non-asyncio BLE package for python, check out e.g. pygattlib, pygatt, bluepy or pywarble instead of bleak.

dhalbert, 2020-08-29:
It is possible to wrap the async calls in such a way that you wait for their results. We used bleak internally in a wrapper library to emulate an existing library that is not async, and is called from non-async code: https://github.com/adafruit/Adafruit_Blinka_bleio/

The `await_bleak()` method here does the waiting: https://github.com/adafruit/Adafruit_Blinka_bleio/blob/94e25c9556f85db8b2d9d57dbb55e2e26cf472e9/_bleio/adapter_.py#L110. You can search for uses of that routine in the rest of the code.

However, as @hbldh says, bleak is designed to be used as `async`, so if you are writing new code, it's best to start that way.

khawajamechatronics, 2020-08-29:
This library is awesome as it is supported on all  OS platform, But I am struggling to understand the flow of the code. Should I post all my questions here or Can we discussion of Email or Skype?

hbldh, 2020-08-29:
I wrote Bleak partly as an exercise to get acquainted and learn the `asyncio` standard module. It takes time to learn if you haven't e.g. done asynchronous programming in other languages.

This is not the forum to ask questions on how to write your program. Likewise, I have not got the time to help you write it. Maybe ask your questions on Stack Overflow? But I think they will point you to learning resources for Python...

khawajamechatronics, 2020-08-30:
I appreciate your work! I have spent some time to understand async io and I am feeling comfortable now with the code flow.

I still needs some more help with your coding/function.  

In the notification code, I wise to use separate UUID for RX and TX. As far in that code RX function has been performed. I have read your functions documentation but still confused how can I add TX function. Can you give some short code to do that function?

Thanks

hbldh, 2020-09-02:
No, I am afraid. I have never done that kind of BLE coding so I cannot help you with that.