hbldh/bleak #267: Trouble Connecting with BLE Device

Reconfirefly, 2020-08-10:
* bleak version:
* Python version:
* Operating System:
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

I'm trying to connect and read from a BLE pressure sensor on my windows 10 machine. I am able to connect the device to the built in bluetooth receiver in the laptop, but when I run a simple connect and read script from Bleak, I get no such luck.

The device shows up as connected in the device manager and bluetooth settings, but it cannot be detected by ```BleakScanner.discover()```. Moreover, the device cannot be connected to by it's address. I have formatted the address with colons, i.e. e8:3a:xx:c9:xx:e3, even though it is given as e83axxc9xxe3. I've edited out a few characters with x's.

I'm hoping I've got the right address for the device. I'm lost at how to continue, and would appreciate any more input. The spec sheet for the device I am trying to connect is here, and the UUID and such is at the bottom of the document.

[Spec Sheet](https://www.te.com/commerce/DocumentDelivery/DDEController?Action=showdoc&DocId=Data+Sheet%7FM5600%7FB11%7Fpdf%7FEnglish%7FENG_DS_M5600_B11.pdf%7FCAT-PTT0063)

CODE:

```
import asyncio
from bleak import BleakClient

address = "e8:3a:xx:c9:xx:e3"
MODEL_NBR_UUID = "F000FA02-0451-4000-B000-000000000000"

async def run(address, loop):
    async with BleakClient(address, loop=loop) as client:
        model_number = await client.read_gatt_char(MODEL_NBR_UUID)
        print("Model Number: {0}".format("".join(map(chr, model_number))))

loop = asyncio.get_event_loop()
loop.run_until_complete(run(address, loop))
```
TRACEBACK:

```
Traceback (most recent call last):
  File "C:/Users/me/PycharmProjects/bleak_test/bleak_1.py", line 13, in <module>
    loop.run_until_complete(run(address, loop))
  File "C:\Users\me\AppData\Local\Programs\Python\Python38-32\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:/Users/me/PycharmProjects/bleak_test/bleak_1.py", line 8, in run
    async with BleakClient(address, loop=loop) as client:
  File "C:\Users\me\PycharmProjects\bleak_test\venv\lib\site-packages\bleak\backends\client.py", line 49, in __aenter__
    await self.connect()
  File "C:\Users\me\PycharmProjects\bleak_test\venv\lib\site-packages\bleak\backends\dotnet\client.py", line 123, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address e8:3a:xx:c9:xx:e3 was not found.

hbldh, 2020-08-15:
Could you please checkout the `feature/find-device` branch of this repo and try connecting to your device with that version? I have made some changes there that might solve the situation.

hbldh, 2020-09-14:
This dhoudl be colsed with version 0.8.0.

closed 2020-09-14 by hbldh