hbldh/bleak #274: Software caused connection abort on service discovery example

vslzl, 2020-08-19:
* bleak version:0.7.1 
* Python version:3.7.3
* Operating System:Debian 9.13
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.43

I want to connect a nordic nrf52DK Uart example from arm based linux os, I tried Ubuntu 16.4 and finally at debian 9.13 but still getting same error on services_discovery.py example.

discover.py works well

but services discovery fails with

> bleak.exc.BleakError: org.bluez.Error.Failed: Software caused connection abort

if I try to connect with bluetoothctl after 'connect CA:xx:xx:xx:xx:xx' command I get all the information

> [NEW] Device CA:xx:xx:xx:xx:xx Nordic_UART
> [bluetooth]# connect CA:xx:xx:xx:xx:xx
> Attempting to connect to CA:56:C9:07:54:CF
> [CHG] Device CA:xx:xx:xx:xx:xxConnected: yes
> Connection successful
> [NEW] Primary Service
>         /org/bluez/hci0/dev_CA_xx_xx_xx_xx_xx/service000a
>         uuid was here
>         Generic Attribute Profile
> [NEW] Primary Service
>         /org/bluez/hci0/dev_CA_xx_xx_xx_xx_xx/service000b
>         uuid was here
>         Vendor specific
> [NEW] Characteristic
>         /org/bluez/hci0/dev_CA_xx_xx_xx_xx_xx/service000b/char000c
>         uuid was here
>         Vendor specific
> [NEW] Characteristic
>         /org/bluez/hci0/dev_CA_xx_xx_xx_xx_xx/service000b/char000e
>         uuid was here
>         Vendor specific
> [NEW] Descriptor
>         /org/bluez/hci0/dev_CA_xx_xx_xx_xx_xx/service000b/char000e/desc0010
>         uuid was here
>         Client Characteristic Configuration
> [CHG] Device CA:xx:xx:xx:xx:xx UUIDs: uuid was here
> [CHG] Device CA:xx:xx:xx:xx:xx UUIDs: uuid was here
> [CHG] Device CA:xx:xx:xx:xx:xx UUIDs: uuid was here
> [CHG] Device CA:xx:xx:xx:xx:xx ServicesResolved: yes
> 

I want to find a uart services uuid and send proper data by using that service. Nordic side is already implemented and bleak examples works on Ubuntu 18.4, win10 PC's but in Debian 9.13 and Ubuntu 16.4 for arm64 same code does not works. 

### What I Did
I tried to change dbus verson on my ubuntu 16.4 and debian 9.13 which was 1.10.xx to 1.12.xx which is same as ubuntu 18.4 Laptop but I cannot make it done. so basically it is default 1.10.x version of dbus here. other dependencies are same on both platforms.

To clarify that, I cannot use bleak service discovery on embedded pc like orange pi 3 in my case with arm64 based cpu. I have no problem at all on laptops which runs the same examples with Ubuntu 18.4 and win10. but in arm64 arch with ubuntu 16.4 and debian 9.13 I got mentioned error.

```
root@orangepi3:~/Ble/bleak/examples# python3 service_explorer.py
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 153, in connect
    ).asFuture(self.loop)
txdbus.error.RemoteError: org.bluez.Error.Failed: Software caused connection abort

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "service_explorer.py", line 69, in <module>
    loop.run_until_complete(run(address, loop, True))
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "service_explorer.py", line 30, in run
    async with BleakClient(address, loop=loop) as client:
  File "/usr/local/lib/python3.7/site-packages/bleak/backends/client.py", line 49, in __aenter__
    await self.connect()
  File "/usr/local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 164, in connect
    raise BleakError(str(e))
bleak.exc.BleakError: org.bluez.Error.Failed: Software caused connection abort
root@orangepi3:~/Ble/bleak/examples# python3 discover.py
CA:xx:xx:xx:xx:xx: Nordic_UART

```

### Finally
Any suggestions will be appreciated. We want to connect, discover the services then send and receive data by using ble at arm64 arch platform device. But we couldn't make it by now.

dlech, 2020-08-19:
I think Bluetooth Low Energy was still somewhat experimental in Ubuntu 16.04 and Debian 9 and you may have to run `bluetoothd` with the `--experimental` flag to enable some features.

You can also check kernel messages with `dmesg` and monitor Bluetooth `btmon` (requires root privileges) to look for other clues.

vslzl, 2020-08-21:
actually @dlech thank you for your help.

First of all the bluez version is not upgradable on the previous OS's I switched to the latest armbian focal version. Now I have BlueZ 5.53 and dbus 1.12.20. But now, it says 

> Exception: Bluetooth adapter not found

I believe that if I can somehow manage to fix that issue it will be ok, the previous issue will never be raised again. Because everything else is same as my ubuntu 18.4 machine.

Also I've checked btmon and the output is:

> root@orangepi3:~# btmon
> Bluetooth monitor ver 5.53
> = Note: Linux version 5.4.45-sunxi64 (aarch64)                         0.290173
> = Note: Bluetooth subsystem version 2.22                               0.290184
> = New Index: 43:xx:xx:xx:xx:xx (Primary,UART,hci1)              [hci1] 0.290188
> = New Index: 00:00:00:00:00:00 (Primary,SDIO,hci0)              [hci0] 0.290191
> @ MGMT Open: bluetoothd (privileged) version 1.14             {0x0001} 0.290196
> @ MGMT Open: btmon (privileged) version 1.14                  {0x0002} 0.290301
>

vslzl, 2020-08-21:
The exception above raised from bleak is from

[this backend file](https://github.com/hbldh/bleak/blob/master/bleak/backends/bluezdbus/discovery.py)

```
def _filter_on_adapter(objs, pattern="hci0"):
    for path, interfaces in objs.items():
        adapter = interfaces.get("org.bluez.Adapter1")
        if adapter is None:
            continue

        if not pattern or pattern == adapter["Address"] or path.endswith(pattern):
            return path, interfaces

    raise Exception("Bluetooth adapter not found")
```

and I think from where this function called, called with optional pattern parameter with None so it fallbacks to hci0 and as my btmon output above, hci0 mac adress is some nulls. it should be hci1 in my case. How can I invoke this function with hci1?

hbldh, 2020-09-02:
The correct solution here is to use the `device` keyword, both in discover, client init and/or connect methods. Closing this.

Documentation about `device` wil be improved in #266.

closed 2020-09-02 by hbldh