hbldh/bleak #277: Missing documentation on how to combine functionalities

jackjansen, 2020-08-30:
* bleak version:
* Python version:
* Operating System:
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

The wait bleak depends on asyncio is not clear from the documentation, and this makes it difficult to write anything slightly more complicated than the examples (unless you know how bleak/asyncio work together).

For example, I want to extend the discovery example with code that tests a discovered device for the availability of a certain service UUID, and then for the devices that support that service UUID get the value of a certain characteristic.

But all the examples are single asyncio loops, so I have absolutely no idea how to structure my code.

Even a single more complex example would probably already do the trick.

### What I Did

Read the documentation and the examples.

hbldh, 2020-09-02:
All examples are single event loops because it is best to stick to one event loop. There are so many usecases for BLE so it is kind of a moot point to try to write example code for all possible ideas that people get. But I agree, the documentation is something I would like to improve: see #266. It requires time I haven't got right now though.

Plan of attack: Your example usecase needs to discover and when scanning is done (cannot connect while scanning), filter on devices with desired service uuid in the [`device.metadata["uuids"]`](https://github.com/hbldh/bleak/blob/3c948ef50e71b91d5a4ce6964d0ebeb09b016720/bleak/backends/device.py#L30) list, connect to each of the desired devices (cannot read chars without connecting) and read the gatt char.

jackjansen, 2020-09-04:
@hbldh I understand the reason for a single event loop. So I presume that the standard/preferred way to tackle an issue like mine is either
- first run a loop to collect all devices and remember them, then a loop to test them, then a loop to get the characteristic value, or
- some sort of work queue where the runloop does the next bit of work, and possibly drops new work into the queue,
- or maybe something completely different:-)

I think your "plan of attack" is to follow the first pattern, correct? I'll do that. If I'm not terribly ashamed of the code I'll attach it here so you can see whether it's useful as an example.

hbldh, 2020-09-10:
@jackjansen Yes, try to keep the it as simple as possible to start with. Something like this:

```python
import asyncio
from bleak import BleakClient, discover

sought_sevice_uuid = "ENTER YOUR SERVICE UUID HERE"

async def run():
    data = {}
    devices = await discover()
    filtered_devices = [d for d in devices if sought_sevice_uuid in d.metadata["uuids"]]
    for d in filtered_devices:
        async with BleakClient(d.address) as client:
            data_from_device = await client.read_gatt_char(
                "ENTER YOUR CHARACTERISTIC UUID OR HANDLE HERE"
            )
            data[d.address] = data_from_device
    print(data)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```
Make sure your uuids are entered with the same casing as Bleak outputs them in `d.metadata["uuids"]`.

I haven't tried the code above s I can't guarantee that it will work, but it is somewhere to start.

hbldh, 2020-09-25:
This issue seems to have no actionable content left. Will close for now and defer example writing to #266.

closed 2020-09-25 by hbldh