hbldh/bleak #281: How do I get full advertisementData

jackjansen, 2020-09-05:
* bleak version:
* Python version:
* Operating System:
MaxOS 
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

I am trying to get at the full advertisementData, i.e. both the info returned from the passive scan as well as the info from the active scan (the first and second callbacks to `centralManager_didDiscoverPeripheral_advertisementData_RSSI_` in case of CoreBluetooth). But from reading the source I get the impression that `BLEDeviceCoreBluetooth` stores only the service UUIDs and the manufacturerData.

There does seem to be special-case code the store the name, but only if the name was included in the passive scan data, if the name was in the scan response (which it happens to be the case for the devices I'm trying to scan) it is happily ignored.

### What I Did

Read the documentation, read the source code.

jackjansen, 2020-09-05:
Actually, just discovered that not only the name is missed if it is in the scan response, but also that the UUIDs advertised in the scan response (if any) overwrite the UUIDs (if any) that were in the passive scan message. I will provide a pull request.

jackjansen, 2020-09-05:
I have created the PR (above).

Also, I have discovered `CentralManagerDelegate.callbacks`, which would allow me to get all advertisement data, but I am not sure whether I should consider that part of the official API.

hbldh, 2020-09-10:
Thank you for this. I will review the PR as soon as I can.

The `CentralManagerDelegate.callbacks` is already used by Bleak and you can hook up your own callback if you use the `BleakScanner` instead of the regular `discover` method. Check [the last example here](https://bleak.readthedocs.io/en/latest/scanning.html#bleakscanner) or the `_find_device_by_address` implementation:
https://github.com/hbldh/bleak/blob/develop/bleak/backends/corebluetooth/scanner.py#L123 and https://github.com/hbldh/bleak/blob/develop/bleak/backends/scanner.py#L90.

hbldh, 2020-09-25:
Regarding this as solved. Will close for now. Reopen if you disagree or open a new issue with the new information.

closed 2020-09-25 by hbldh