hbldh/bleak #284: Finding the MAC address on macOS

bardiabarabadi, 2020-09-09:
* bleak version: 0.7.1
* Python version: 3.8
* Operating System: macOS Catalina 10.15.6

### Description

I managed to get a list of BLE devices using the same procedure as the scanner.py, i.e. the `discover()` method. Since I am using bleak on macOS, the returned device type is `BLEDeviceCoreBluetooth` which has some attributes including `address`.

The problem is that according to the documentation, the address field when using the macOS backend is actually the CBPeripheral identifier (and not the MAC address). Of course, this is not an issue by itself, but when I want to connect to the device using the `BleakClient` class, I need to have the MAC address (not the identifier).

Apparently, it is not possible to obtain the MAC address from the identifier (it changes every time).

Can you please tell me how can I get the MAC address, or alternatively use the identifier to connect to the BLE device via `BleakClient` class?

Thank you.

dlech, 2020-09-09:
Please see #140. Apple does not expose the Bluetooth address for security/privacy reasons.

bardiabarabadi, 2020-09-09:
Thanks for the response @dlech 
Is there any way to connect to the BLE device using the `BleakClient` class on macOS (considering the fact that the Bluetooth address is not available?)

dlech, 2020-09-09:
If you are using the `discover()` function, then just call the `connect()` method on the device that was discovered.

bardiabarabadi, 2020-09-09:
Yes, I am using the `discover()` function. The only `connect()` method I found is in bleak/bleak/backends/corebluetooth/client.py in the `BleakClientCoreBluetooth` class. However, `BleakClientCoreBluetooth` requires an address to initialize (that is not available). Is this the `connect()` function you are referring to? If not, can you show me a quick example of how to import and use it?

dlech, 2020-09-09:
Right. So you can use the `address` attribute of the discovered device to create the client object.

```python
import asyncio
from bleak import discover

async def run():
    devices = await discover()
    for d in devices:
        print(d)
        client = BleakClient(d.address)
        await client.connect()
        # do something with client

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```

bardiabarabadi, 2020-09-09:
Thank you for the code. That was actually the first thing I tried before opening this case. However, I rewrote the code to make sure that I am not missing anything and I got the same error (Device with address xxxxxx-xx-xxx-xxxxx was not found)

here is my code and the error I get:
```
import asyncio
from bleak import discover, BleakClient

async def find_muses():
    devices = await discover()
    muses = []
    print('found ' + str(len(devices)))
    for d in devices:
        if 'Muse' in d.name:
            print(d)
            muses.append(d)
            client = BleakClient(d.address)
            await client.connect()
    return muses

if __name__ == '__main__':
    loop = asyncio.get_event_loop()
    muses = loop.run_until_complete(find_muses())
```

Output:
```
found 20
55C835D6-B712-4109-8AFD-6760C227F601: Muse-C3DD
Traceback (most recent call last):
  File "/Users/bardiabarabadi/PycharmProjects/bleakTest/tester.py", line 26, in <module>
    muses = loop.run_until_complete(find_muses())
  File "/Users/bardiabarabadi/opt/miniconda3/envs/condaMuse/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/Users/bardiabarabadi/PycharmProjects/bleakTest/tester.py", line 16, in find_muses
    await client.connect()
  File "/Users/bardiabarabadi/opt/miniconda3/envs/condaMuse/lib/python3.8/site-packages/bleak/backends/corebluetooth/client.py", line 77, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address 55C835D6-B712-4109-8AFD-6760C227F601 was not found
```

dlech, 2020-09-10:
Thanks for your persistence :smile:  I see what is going on now. The `connect()` method in the `BleakClient` object does discovery again. For some reason your device doesn't get discovered the second time. Maybe you could add a longer timeout when calling `connect()`?

I think we could improve things by adding a connect method to the device object that returns a connected `BleakClient` object (and takes an arg for callbacks like suggested in https://github.com/hbldh/bleak/issues/276#issue-687136064). This way we don't have to detect a device twice.

dhalbert, 2020-09-10:
Also see #191.

bardiabarabadi, 2020-09-10:
Hello again. So the issue is gone now (by increasing the timeout) and I have to say that I impressed by your quick and professional responses. Thank you so much for developing and maintaining this library. I probably will get back with more questions, but until then I am going to close this thread. Cheers.

closed 2020-09-10 by bardiabarabadi

dlech, 2022-10-07:
FYI, I have found a hack for this. See #1073. Give it a 👍 if this is still a needed feature.