hbldh/bleak #287: Scan results are not cached

jackjansen, 2020-09-12:
* bleak version: develop
* Python version: 3.8
* Operating System: OSX
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

Related to #286 but different.

### Description

Trying to scan for a number of identical devices (with `discover()` and then manually filtering the results), then connect to each one and perform some operation on it.

The whole sequence takes excruciatingly long, because each device connection will do another fresh scan. And for my use case #286 means that not only will it take excruciatingly long but usually fails.

### What I Did

```
  async def pseudocode():
    devices = await bleak.discover()
    wanted_devices = my_filter(devices)
    for d in wanted_devices:
      async with bleak.BleakClient(device.address) as server:
        services = await server.get_services()
```

Also tried an `async for` with tasks, but that seems to make matters even worse.

### Observations

Scan results can in general be cached, because even a device with a random address will only change its address once in a while (I think 15 minutes or something similar). So if the scan results were cached the connect could simply reuse the results of the previous scan operation, if not too old.

Alternatively, or possibly additionally, if the scanner was an async generator (https://www.python.org/dev/peps/pep-0525/) returning its results as they came in the whole code could be structured much more elegant, something like:

```
  async def pseudocode():
    async for device in bleak.discover_generator():
      if my_filter_one_device(device):
        async with bleak.BleakClient(device.address) as server:
          services = await server.get_services()
```

This is much more in line with the API provided by CoreBluetooth or other modern BLE APIs (but, of course, without all the horrible callbacks that we don't need anymore with asyncio in Python).

dhalbert, 2020-09-12:
Have you seen #285? I think that would help you.

hbldh, 2020-09-14:
I believe that you cannot leave the scanner running while connecting. Thus, ensuring that the async generator works as desired will be hard, and not something I want to be responsible for maintaining as a developer of Bleak. 
I will never build any caching into Bleak if I can avoid it and #279 and #285 provides much better handling than exists in version 0.7.1. Wait until 0.8.0.

closed 2020-09-14 by hbldh