hbldh/bleak #289: set default branch on GitHub to develop?

dlech, 2020-09-14:
I think it would be nice to set the default branch on GitHub to develop instead of master. This makes it easier for those forking and making pull requests to use the right branch for development. Also, it would be easier for users to find the latest CHANGELOG that included the unreleased changes, which is a nice way to know if that bug fix you need has been released yet or not.

hbldh, 2020-09-14:
I agree. It would facilitate a better handling, even though it doesn't portray the true "production" state. Feel free to change it!

dlech, 2020-09-14:
I don't think I have access to change repository settings.

hbldh, 2020-09-14:
Understood. I had hoped that was part of the deal. I will change tomorrow.

hbldh, 2020-09-17:
Done

closed 2020-09-17 by hbldh