hbldh/bleak #292: No "Handle" entry in bluez characteristic dictionary

dhalbert, 2020-09-15:
* bleak version: 0.8.0? (commit aa54b08)
* Python version: 3.8.2
* Operating System: Ubuntu 20.04
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.53
* txdbus: 1.1.1

My notification callbacks are now getting `None` as the first argument, which should be the integer handle. Tracking this down, it appears to be because the characteristic info collected during discovery via txdbus messages, in `BleakClientBlueZDBus.get_services()`, does not contain handles. That is, at https://github.com/hbldh/bleak/blob/aa54b08a7c5a491bd909917bedb3a582dc9929d5/bleak/backends/bluezdbus/client.py#L427-L432
the value of `char` is a dict without a `"Handle"` key. Here is a typical `char` dict:
```python
{'UUID': '0000fff5-0000-1000-8000-00805f9b34fb', 'Service': '/org/bluez/hci0/dev_50_F1_4A_52_AF_9F/service0023', 'Value': [], 'Flags': ['write']}
```
Am I mistaken in thinking there should there be a `"Handle"` key? It looks like you were expecting one, but guarding against the possibility it was missing.

dlech, 2020-09-15:
It looks like "Handle" is a sever-only property, so it is not available to us.. https://github.com/bluez/bluez/blob/master/doc/gatt-api.txt#L275

I think it would be safe to get the characteristic handle from the D-Bus path though. See https://github.com/bluez/bluez/blob/e062b29cf43819f6781e6eafd494ba88f78ea5be/src/gatt-client.c#L1739

The last 4 characters of `charyyyy` are the handle in hex.

hbldh, 2020-09-16:
But why did this happen? I have tested this and seen the ``Handle`` attribute! I would like to say that this was an oversight on my part when testing, but I know I tested this!

Regardless, it needs to be handled. 

@dlech A reason I do not like using the path is the comment at the bottom of #150. I do not fully trust BlueZ to always adhere to these naming conventions, but since you have found the code to support it then it might be fine. I think Bleak already does it actually, at least in the `characteristic.py` file, but not when setting `_char_path_to_handle`. That must be what I actually tested: that a handle could be extracted from the path, not that it actually flowed through to the notifications...

dhalbert, 2020-09-17:
Fixed by #293.

PR #293 (2020-09-16): Fix getting handle in BlueZ backend
The "Handle" property of the org.bluez.GattCharacteristic1 D-Bus interface is server-only, so it isn't available to Bleak. Instead,
we can scrape the handle from the D-Bus path of the characteristic.

Full object path format for reference:

    [variable prefix]/{hci0,hci1,...}/dev_XX_XX_XX_XX_XX_XX/serviceXX/charYYYY

Fixes #292
commit dc1b5a2d (2020-09-16): Fix getting handle in BlueZ backend

The "Handle" property of the org.bluez.GattCharacteristic1 D-Bus
interface is server-only, so it isn't available to Bleak. Instead,
we can scrape the handle from the D-Bus path of the characteristic.

Full object path format for reference:

    [variable prefix]/{hci0,hci1,...}/dev_XX_XX_XX_XX_XX_XX/serviceXX/charYYYY

Fixes #292
closed 2020-09-17 by dhalbert