hbldh/bleak #294: 'BleakClientCoreBluetooth' object has no attribute '_disconnected_callback_client'

dhalbert, 2020-09-16:
* bleak version: 57e35147a
* Operating System: MacOS 10.15.6

@dlech The disconnected callback code for MacOS seems to be missing setting an attribute. I tried one of my usual examples immediately after merging #291, and got this error. I don't see this issue on bluez.
```
  ...
  File "/Users/halbert/repos/Adafruit_Blinka_bleio/_bleio/adapter_.py", line 359, in _connect_async
    await client.connect(timeout=timeout)
  File "/Users/halbert/repos/hbldh/bleak/bleak/backends/corebluetooth/client.py", line 88, in connect
    manager.disconnected_callback = self._disconnected_callback_client
AttributeError: 'BleakClientCoreBluetooth' object has no attribute '_disconnected_callback_client'
```

dlech, 2020-09-16:
Sorry about that (careless mistake on my part). One fix coming right up.

dhalbert, 2020-09-16:
Fixed by #295

PR #295 (2020-09-16): Fix deleting wrong method in CoreBluetooth backend
In 666395e26b8a, `_disconnected_callback_client()` was accidentally deleted instead of `set_disconnected_callback()`. This restores the former and deletes the latter.

Fixes #294
commit 800510cd (2020-09-16): Fix deleting wrong method in CoreBluetooth backend

In 666395e26b8a, _disconnected_callback_client() was accidentally
deleted instead of set_disconnected_callback(). This restores the
former and deletes the latter.

Fixes #294
closed 2020-09-16 by dhalbert