hbldh/bleak #296: Bleak and Applications with GUI

bardiabarabadi, 2020-09-17:
* bleak version: 0.7.1
* Python version: 3.8
* Operating System: macOS Catalina 10.15.6

### Description

So before anything, I know that this "issue" that I am writing is not actually Bleak's issue/bug. It is more of a question than a bug report. In summary, I want to use Bleak in an application that has a graphical user interface and I am having issues with the asynchronous nature of Bleak. I explained the steps I've taken and the errors I faced in detail below:

### What I Did

Since I already had a GUI written in Kivy, I tried to use Bleak and integrate it into the Kivy's Button callbacks.

For the search (discovery) part, I wrote `async def search_for_bles()` in which I call `await discover()`. To call this function from the button's callback I had to use `asyncio.run()` method, which worked fine! 

I used the same strategy (`await`, `async def`, and `asyncio.run`) `connect()`, here is a part of the code (I removed the pointless lines)

```
def connect_btn_callback(self,......):

    self.ble_obj=BLE(found_client, ...... )

    asyncio.run(self.ble_obj.connect()) # [1] Works fine.    
    asyncio.run(self.ble_obj.ask_control()) # [2] Runtime Error: 'Event loop is closed'

``` 

in the BLE class:
```
def __init__ (self, client, .....):
    self.client=client
    ....

async def connect (self):
    await self.client.connect() # [3] works fine
    await self.ask_control() # [4] works fine

async def ask_control(self):
    await self.client.write_gatt_char(constants.CONTROL_HANDLE, bytearray([0x02, 0x73, 0x0a]), False)

```

The comments in the code are showing where the error occurs (this also happens for disconnect, and other methods)

What is interesting for me is that the `ask_control` inside [1], i.e. [4], works without an issue (this means that the `ask_control` function is healthy). However, [2] gives me an error, `RunTime error: 'event loop is closed'`

### Other things that I have tried:
After a few days working on this issues full-time, I found many related issues but none of the solutions worked:
1. `from asgiref.sync import async_to_sync`
2. Using `loop=asyncio.get_event_loop()`
3. Using `loop=asyncio.get_running_loop()`
4. Using `loop=asyncio.new_event_loop` and `asyncio.set_event_loop(loop)` 

### What I am asking
So I am presenting this issue to you not (especifically) as the developers of Bleak, but as experienced python users/developers. I want to have bleak and a cross-platform GUI in one app. Even if you think that using another GUI makes using Bleak easier, please let me know. 

Thank you.

hbldh, 2020-09-17:
I think you cannot use `asyncio.run` to handle bleak; you have to keep it alive in the event loop.

Have you read through [Kivys documentation on asyncio](https://kivy.readthedocs.io/en/master/api-kivy.app.html#asynchronous-app)? You need to run the app through asyncio and let Bleak share the same event loop as the application.

Carglglz, 2020-09-17:
@bardiabarabadi 

> The comments in the code are showing where the error occurs (this also happens for the disconnect, and other methods)
What is interesting for me is that the ask_control inside [1], i.e. [4], works without an issue (this means that the ask_control function is healthy). However, [2] gives me an error, RunTime error: 'event loop is closed'

The reason for this is explained in the `asyncio.run` docstring. 
The important part:

```
This function always creates a new event loop and closes it at the end.
It should be used as a main entry point for asyncio programs, and should
ideally only be called once.
```
(do `help(asyncio.run)` to see the complete version)

So I use `loop.run_until_complete` instead and it works great. (at least with PyQt5) 👍

bardiabarabadi, 2020-09-18:
Thank you, @hbldh for your response. So the reason I didn't use `async_run()` (Kivy) was that it was only available on the `rc` versions (not the stable 1.11.1). Anyway, I updated Kivy to 2.0.0rc3 and tried running the Bleak functions using `loop=asyncio.get_event_loop()` and `loop.run_until_complete`. Of course, another problem arose; apparently `run_until_complete` closes the `loop` after the future function returns, and because the loop is shared with UI, the applictation crashes. 

For future reference: the issue can be solved by having these two lines at the beginning of the code:
```
import nest_asyncio
nest_asyncio.apply()
```
Thank you for answering my question, everything is working fine now. I am closing this thread. 

P.S. @Carglglz Thanks for your comment. As I mentioned in the "Other things that I have tried" section of my initial post, getting the currently running loop (with `asyncio.get_event_loop()`) gave me the same error. The reason is that the Kivy does not lwt go of the running thread and thus there is not any event loop for asyncio to extract.

closed 2020-09-18 by bardiabarabadi

jimakos96, 2022-03-23:
Hello , i am running the kivy example and i want for testing to just make a  connection with a specific device and read data from a characteristic .My issue is that  i want to make one single connection and keep receiving data from the callback handler but if i remove the (while True:)  ,then the co routine example() ends immediately .Is there any way to have the kivy running along the co-routine without the endless loop?

`from kivy.app import App

# from kivy.core.window import Window
from kivy.uix.label import Label
from kivy.uix.scrollview import ScrollView

# bind bleak's python logger into kivy's logger before importing python module using logging
from kivy.logger import Logger
import logging

logging.Logger.manager.root = Logger

import asyncio
import bleak

ADDRESS = "D9:E2:A3:44:23:2D"
hr_char ="00002a37-0000-1000-8000-00805f9b34fb"

class ExampleApp(App):
    def __init__(self):
        super().__init__()
        self.label = None
        self.running = True

    def build(self):
        self.scrollview = ScrollView(do_scroll_x=False, scroll_type=["bars", "content"])
        self.label = Label(font_size="10sp")
        self.scrollview.add_widget(self.label)
        return self.scrollview

    def line(self, text, empty=False):
        Logger.info("example:" + text)
        if self.label is None:
            return
        text += "\n"
        if empty:
            self.label.text = text
        else:
            self.label.text += text

    def on_stop(self):
        self.running = False

    def callback(self,sender, data):
        self.line(f"{str(data[-1])} bpm")

    async def example(self):

        while True:
            try:
                    async with bleak.BleakClient(ADDRESS) as client:
                        self.line(f"Connected: {client.is_connected}")
                        await client.start_notify(hr_char, self.callback)

            except bleak.exc.BleakError as e:
                        self.line(f"  error {e}")
                        asyncio.sleep(10)

async def main(app):
    await asyncio.gather(app.async_run("asyncio"), app.example())

if __name__ == "__main__":
    Logger.setLevel(logging.DEBUG)

    # app running on one thread with two async coroutines
    app = ExampleApp()
    asyncio.run(main(app))
`

dlech, 2022-03-23:
A simple way to wait forever is `await asyncio.Event().wait()`.