hbldh/bleak #297: Sending lists of data

Darynbright, 2020-09-17:
* bleak version: 0.7.1
* Python version: 3.8.5
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

I am trying to send a bunch of data via an Arduino Nano 33 BLE. I have recorded the data and then stored the data on the micro. I want to then send the list across via BLE. The issue I'm having is that the Bleak module tends to take a various amount of time between connecting to the peripheral and between reading the characteristics via start_notify. I want to be able to say, once connected, send the data only when Bleak is reading the characteristics but trying to use a delay of some kind. From Arduino's side, there doesn't seem to be a 'when characteristic is read' notification, rather a successfully sent notification. It is sending the data without being read for the beginning very (milli)seconds.

Is there a way of accurately waiting a certain amount of time, every time, to then send the data at the correct time? I'm not sure if this is a Bleak or Arduino issue though.

### What I Did

Arduino Code:
void loop() {
  while (1) {
    // wait for a BLE central
    BLEDevice central = BLE.central();
    
    // if a central is connected to the peripheral:
    if (central.connected()) {
      int index = 0;
      int sent = 0;
      while(central.connected()){
        // send '0' while waiting a certain length of time
        if(index < index_value){
          x_acc_char.writeValue(0);
        }
        else{
          x_acc_char.writeValue(data[sent]);
          sent ++;
          //Print number of times the data is sent
          Serial.println(sent);
        }
        //Print total iterations of while loop
        Serial.println(index);
        index ++;
      }
}

Python Code:

def notification_handler_1(sender, data):
    """Simple notification handler which converts the data, stores it and then prints the data received with a timestamp."""
    value = convert_byte_array_to_decimal(data)
    first_list.append(value)
    print(datetime.now(), value)

async def run_1(address, debug=False):
    if debug:
        import sys

        # loop.set_debug(True)
        l = logging.getLogger("asyncio")
        l.setLevel(logging.DEBUG)
        h = logging.StreamHandler(sys.stdout)
        h.setLevel(logging.DEBUG)
        l.addHandler(h)
        logger.addHandler(h)

    async with BleakClient(address) as client:
        x = await client.is_connected()
        logger.info("Connected: {0}".format(x))
        await client.start_notify(char_uuid("2101"), notification_handler_1)
        await asyncio.sleep(7.5)
        await client.stop_notify(char_uuid("2101"))

Output:

First running of program:
Last couple lines of output from python: 
2020-09-17 10:29:54.274302 0
2020-09-17 10:29:54.299279 0
2020-09-17 10:29:54.332273 0
2020-09-17 10:29:54.392772 0
2020-09-17 10:29:54.393770 0

Same timestamp from Arduino:
10:29:54.293 -> 13591  //Index
10:29:54.293 -> 1493    //Sent
10:29:54.293 -> 13592  //Index ....
10:29:54.293 -> 1494
10:29:54.293 -> 13593
10:29:54.430 -> 1495

When the program is rerun:
Last couple lines of output from python: 
2020-09-17 10:32:53.317085 0
2020-09-17 10:32:53.375052 0
2020-09-17 10:32:53.375052 0
2020-09-17 10:32:53.379051 0
2020-09-17 10:32:53.434971 0

Same timestamp from Arduino:
10:32:53.388 -> 13058  //Index
10:32:53.388 -> 960      //Sent
10:32:53.388 -> 13059  //Index ....
10:32:53.388 -> 961
10:32:53.388 -> 13060
10:32:53.422 -> 962

There is no one time that Bleak seems to take to start reading the data.

hbldh, 2020-09-25:
I am not sure what to do with this. I see what you expect to happen, but I do not know why it happens. Bleak notifications work just fine with all devices I have tried it with, yielding exactly the value that I want. 

There are two possibilities here as I see it: 

1. your Arduino code does not do what you want it to do. How many zeros are sent before `index > index_value`. Bleak will handle notifications sequentially as far as I have seen, and if you send a lot of zero traffic first, then it might seem like you never get any real data?
2. your method `convert_byte_array_to_decimal` does not do what you expect it to do?

Try this out, because I do not think this is a Bleak issue.

Darynbright, 2020-10-01:
I managed to find a solution. I feel like it is just a bit of a cheat method though. It takes the time delay from the micro when it receives a packet and then starts sending the list.

while(central.connected()){
      current_time = millis();
      if(current_time <= past_time + 4){
        if(start == 0){
          x_acc_char.writeValue(0);
        }
        else{
          x_acc_char.writeValue(data[sent]);
          sent ++;     
        }
        times_time ++;
        past_time = current_time;
      }
      else{
        if((times_time >= 4) && (start == 0)){
          x_acc_char.writeValue(0);
        }
        else{
          start = 1;
          x_acc_char.writeValue(data[sent]);
          sent ++; 
        }
        times_time = 0;
        past_time = current_time;
      }
      delay(1);
    }

It seems like I'm still only getting a data rate of 40Hz. I tried to remove the need for the Arduino to receive 'acknowledgement' packets which increased the frequency of the arduino output by a lot, but Bleak still maintained it's 40Hz input. You have mentioned that you got a data rate of 500Hz previously. Was that through the standard code, or did you have to adjust some things?

hbldh, 2020-10-06:
It was trhough standard code, but when I looked closer on the external package I used for communicating to the peripheral, it records on the device in 500Hz but transmits bundles of data points at a lower rate, around 100 Hz. So no, I have then not transmitted at such high rates.

I think the acknowledgments are the key to getting what you want here. You need to implemement an Indicate-capable characteristic instead of a Notify-capable, i.e. make sure you get a acknowledgement from the BleakClient that the data has been received.

Darynbright, 2020-10-06:
From what I've read about Notify vs Indicate characteristics, Indicate requires acknowledgement that the central has received the data. Would that not slow down the transmission rates though?

hbldh, 2020-10-06:
It might. But if it is either fast data or correct data, I know which one I would choose.

closed 2021-10-06 by dlech