hbldh/bleak #298: hello,await client.start_notify(client,notify),this is a async function,how  to handle it synchronized?

tomuGo, 2020-09-19:
sorry about my english.
`    async def enable_notif(self, client):
        await client.start_notify("0000ffe4-0000-1000-8000-00805f9b34fb", self.notification_handler)`

the bluedevice send byte one by one to me, I should deal with it one by one.but this function is async.how  to handle it synchronized?
thanks

hbldh, 2020-09-19:
The `data` variable is a bytearray, so when you loop over it you enqueue one byte at a time. Do
```python

def notification_handler(self,sender, data):
    logger.debug("收到息："+data.hex())
    # 写队列
    self.queue.enQueue(data)
    self.DecodeQueue.decodeData(self.queue,self.signals)
```

instead.

tomuGo, 2020-09-20:
> Do

im sorry .There are some misleading in code,there is no for loop in code .
```
async def enable_notif(self, client):
    await client.start_notify("0000ffe4-0000-1000-8000-00805f9b34fb", self.notification_handler)

def notification_handler(self,sender, data):
    logger.debug("收到信息："+data.hex())
    # 写队列
    self.queue.enQueue(data)
    elf.DecodeQueue.decodeData(self.queue,self.signals)
```
bluedevice send one byte to me,not bytearray.i need function  notification_handler() synchronized

hbldh, 2020-09-21:
> i need function notification_handler() synchronized.

I do not understand your question, or what you are really after. It is a synchronous function, called as a callback from the OS. If data arrives one by one from your device, it is your device's problem, not Bleak's. Do you really need a queue?

hbldh, 2020-09-25:
Will close this for now. I do not know if I can do anything here.

closed 2020-09-25 by hbldh