hbldh/bleak #299: GATT Server connected but cannot read services

HasanFarag, 2020-09-20:
* bleak version: 0.7.1
* Python version: 3.8
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description

I wanted to check the services of GATT server. it worked once but afterthat i got that error. i restarted the server but still the same error. it is connected but cannot read GATT services

### What I Did
import platform
import asyncio
import logging
from bleak import BleakClient

async def run(address, debug=False):
    log = logging.getLogger(__name__)
    if debug:
        import sys

        log.setLevel(logging.DEBUG)
        h = logging.StreamHandler(sys.stdout)
        h.setLevel(logging.DEBUG)
        log.addHandler(h)

    async with BleakClient(address) as client:
        x = await client.is_connected()
        log.info("Connected: {0}".format(x))

        for service in client.services:
            log.info("[Service] {0}: {1}".format(service.uuid, service.description))

```Traceback (most recent call last):
  File "test.py", line 69, in <module>
    loop.run_until_complete(run(address, True))
  File "C:\Users\jmtro\AppData\Local\Programs\Python\Python38-32\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "test.py", line 28, in run
    async with BleakClient(address) as client:
  File "C:\Users\jmtro\AppData\Local\Programs\Python\Python38-32\lib\site-packages\bleak\backends\client.py", line 49, in __aenter__
    await self.connect()
  File "C:\Users\jmtro\AppData\Local\Programs\Python\Python38-32\lib\site-packages\bleak\backends\dotnet\client.py", line 150, in connect
    services = await self.get_services()
  File "C:\Users\jmtro\AppData\Local\Programs\Python\Python38-32\lib\site-packages\bleak\backends\dotnet\client.py", line 249, in get_services
    raise BleakDotNetTaskError(
bleak.exc.BleakDotNetTaskError: Could not get GATT services: Unreachable
```

hbldh, 2020-09-21:
Strange. The `Unreachable` status is originating in the Windows backend, not in Bleak, so I cannot know why this is happening. What Windows version (1909? 2004?) do you run?

HasanFarag, 2020-09-21:
2004

hbldh, 2020-09-21:
2004 has been a cause of problems for other users. But if you have problems in both Windows AND Linux, it sounds like your device is a prime suspect, not bleak. Have you tried other ways of connecting to it?

hbldh, 2020-10-20:
I have nothing further to add to this issue. Will close for now.

closed 2020-10-20 by hbldh