hbldh/bleak #300: cx_Freeze complains of missing modules; executable will not run when Bleak is imported

NbrTDB, 2020-09-21:
* bleak version: 0.7.1
* Python version: 3.8.5
* cx_Freeze version: 6.2
* Operating System: Windows 10 2004, build 19041.508

### Description

When using cx_Freeze to build a simple executable importing bleak , warnings about various missing modules are generated and the built executable fails to load. The variety of the missing modules is such that I'm really not sure where to start on fixing this, so I figured I'd ask if anyone else has experience with cx_Freeze and bleak and has encountered this before.

I understand this may wind up being more of a cx_Freeze issue than a bleak issue, so I may have to ask the question over there.

When the same python file (test.py) is run directly, everything works fine:
```Batch
py -3.8 test.py
(after a few seconds, produces list of BLE devices here, as expected)
Finished!
```
Yet the powershell output when executable is run is as below: 
```
.\test.exe

Unhandled Exception: System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.ArgumentNullException: Value cannot be null.
Parameter name: key
   at System.ThrowHelper.ThrowArgumentNullException(ExceptionArgument argument)
   at System.Collections.Generic.Dictionary`2.FindEntry(TKey key)
   at System.Collections.Generic.Dictionary`2.TryGetValue(TKey key, TValue& value)
   at Python.Runtime.Runtime.InitializePlatformData()
   at Python.Runtime.Runtime.Initialize(Boolean initSigs)
   at Python.Runtime.PythonEngine.Initialize(IEnumerable`1 args, Boolean setSysArgv, Boolean initSigs)
   at Python.Runtime.PythonEngine.InitExt()
   --- End of inner exception stack trace ---
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   at System.Reflection.RuntimeMethodInfo.UnsafeInvokeInternal(Object obj, Object[] parameters, Object[] arguments)
   at System.Reflection.RuntimeMethodInfo.Invoke(Object obj, BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at System.RuntimeType.InvokeMember(String name, BindingFlags bindingFlags, Binder binder, Object target, Object[] providedArgs, ParameterModifier[] modifiers, CultureInfo culture, String[] namedParams)
   at clrModule.PyInit_clr()
```

### What I Did

setup.py:
```python
import subprocess
import sys
cxf = 'cx_Freeze'

from cx_Freeze import setup, Executable
# suppress tons of terminal output; set "silent" to False if you think there's a build problem
build_exe_options = {"silent": False}

setup(name = "test",
	version = "0.1",
	description = "",
	options = {"build_exe": build_exe_options},
	executables = [Executable("test.py")])
```

test.py:
```python
#!/usr/bin/env python3
import asyncio
from bleak import discover

async def run():
    devices = await discover()
    for d in devices:
        print(d)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
[... 535 lines cut ...]
copying C:\Python38\lib\site-packages\bleak-0.7.1.dist-info\top_level.txt -> build\exe.win-amd64-3.8\lib\bleak-0.7.1.dist-info\top_level.txt
copying C:\Python38\lib\site-packages\bleak-0.7.1.dist-info\WHEEL -> build\exe.win-amd64-3.8\lib\bleak-0.7.1.dist-info\WHEEL
copying C:\Python38\DLLs\_asyncio.pyd -> build\exe.win-amd64-3.8\lib\_asyncio.pyd
copying C:\Python38\DLLs\_bz2.pyd -> build\exe.win-amd64-3.8\lib\_bz2.pyd
copying C:\Python38\DLLs\_ctypes.pyd -> build\exe.win-amd64-3.8\lib\_ctypes.pyd
copying C:\Python38\DLLs\_decimal.pyd -> build\exe.win-amd64-3.8\lib\_decimal.pyd
copying C:\Python38\DLLs\_hashlib.pyd -> build\exe.win-amd64-3.8\lib\_hashlib.pyd
copying C:\Python38\DLLs\_lzma.pyd -> build\exe.win-amd64-3.8\lib\_lzma.pyd
copying C:\Python38\DLLs\_multiprocessing.pyd -> build\exe.win-amd64-3.8\lib\_multiprocessing.pyd
copying C:\Python38\DLLs\_overlapped.pyd -> build\exe.win-amd64-3.8\lib\_overlapped.pyd
copying C:\Python38\DLLs\_queue.pyd -> build\exe.win-amd64-3.8\lib\_queue.pyd
copying C:\Python38\DLLs\_socket.pyd -> build\exe.win-amd64-3.8\lib\_socket.pyd
copying C:\Python38\DLLs\_ssl.pyd -> build\exe.win-amd64-3.8\lib\_ssl.pyd
copying C:\Python38\lib\site-packages\clr.pyd -> build\exe.win-amd64-3.8\lib\clr.pyd
copying C:\Python38\DLLs\pyexpat.pyd -> build\exe.win-amd64-3.8\lib\pyexpat.pyd
copying C:\Python38\DLLs\select.pyd -> build\exe.win-amd64-3.8\lib\select.pyd
copying C:\Python38\DLLs\unicodedata.pyd -> build\exe.win-amd64-3.8\lib\unicodedata.pyd
copying C:\Python38\lib\site-packages\win32\win32wnet.pyd -> build\exe.win-amd64-3.8\lib\win32wnet.pyd
copying c:\python38\python38.dll -> build\exe.win-amd64-3.8\lib\python38.dll
copying c:\python38\vcruntime140.dll -> build\exe.win-amd64-3.8\lib\vcruntime140.dll
copying C:\Python38\DLLs\libcrypto-1_1.dll -> build\exe.win-amd64-3.8\lib\libcrypto-1_1.dll
copying C:\Python38\DLLs\libssl-1_1.dll -> build\exe.win-amd64-3.8\lib\libssl-1_1.dll
copying C:\Python38\DLLs\libffi-7.dll -> build\exe.win-amd64-3.8\lib\libffi-7.dll
copying C:\Python38\lib\site-packages\Python.Runtime.dll -> build\exe.win-amd64-3.8\lib\Python.Runtime.dll
```

hbldh, 2020-09-21:
Don't use `cx_freeze` with bleak (see https://github.com/pythonnet/pythonnet/issues/66). Try using pyinstaller instead if you have to.

NbrTDB, 2020-09-22:
I see - thanks for the info! Just to clarify my understanding (and I suppose anyone else who comes across this), is the issue that bleak has external dependencies which `cx_freeze` can't detect/resolve, like with pythonnet? Thus requiring something similar to the PR mentioned in that issue?

I did indeed try pyinstaller and it worked perfectly.

hbldh, 2020-09-22:
Good that it worked. It seems to be some problem with dependencies when freezing `pythonnet`, yes.

closed 2020-09-22 by hbldh