hbldh/bleak #308: client.set_disconnected_callback - callback does not appear to fire on disconnect

NbrTDB, 2020-09-25:
* bleak version: 0.8.0
* Python version: 3.8.5
* Operating System: Windows 10 2004, build 19041.508

### Description

After connecting to a device, I am trying to set up a callback so that I am alerted if/when the device disappears (for instance if it wanders out of range, or waiting for confirmation of disconnect after `await client.disconnect()`.

However, while the device definitely disconnects (be it through range, or being turned off, or `await client.disconnect()`) the callback never runs and no errors are produced either.

I'm not entirely sure if I'm doing this right!

### What I Did

For the purpose of testing, I put these two methods together in the class I'm implementing for BLE stuff; `__callback_test()` is fed the target MAC address and then connects to the device, sets a disconnected_callback, pokes the GATT, then disconnects. 

```python
    def __disconnect_callback(self):
        # fires when a device is disconnected
        print("Disconnect callback...")

    async def __callback_test(self, mac):
        print("Connecting to mac: {0}".format(mac), flush=True)
        client = BleakClient(mac)
        await client.connect()
        # try to set a callback
        client.set_disconnected_callback(self.__disconnect_callback)
        # do something with the connection, blah blah 
        print("Poking services...".format(mac), flush=True)
        await client.get_services()
        # now disconnect        
        print("Trying disconnect...".format(mac), flush=True)
        await client.disconnect()
```

Elsewhere in the class is the code to initiate the test:

```python
                mac = "[MAC address here]"
                loop = asyncio.get_event_loop()
                loop.run_until_complete(self.__callback_test(mac))
```

The console output is as below; the device shows a connection formed, then a few seconds later, disconnected, but the callback doesn't appear to run from the console output. As stated before there is no error and the program is still running after this:
```
Connecting to mac: [MAC address here]
Poking services...
Trying disconnect...
```

hbldh, 2020-09-25:
Your disconnect callback needs to accept one input argument. See docstring with enclosed example: https://github.com/hbldh/bleak/blob/develop/bleak/backends/client.py#L68-L89

NbrTDB, 2020-09-25:
Thanks for the quick response! I had amended the callback to the following, but the behaviour is the same:

```python3
    def __disconnect_callback(self, client):
        # fires when a device is disconnected
        print("Disconnected from {0}".format(client.address))
```

I also rearranged the test method so the callback was set before connection,  but still to no avail:

```python3
    async def __callback_test(self, mac):
        print("Connecting to mac: {0}".format(mac), flush=True)
        client = BleakClient(mac)
        client.set_disconnected_callback(self.__disconnect_callback)
        await client.connect()
        # do something with the connection, blah blah 
        print("Poking services...".format(mac), flush=True)
        await client.get_services()
        # now disconnect        
        print("Trying disconnect...".format(mac), flush=True)
        await client.disconnect()
```

hbldh, 2020-09-25:
This works fine for me:
```python
def __disconnect_callback(client):
    try:
        # fires when a device is disconnected
        print("Disconnected from {0}".format(client.address))
    except Exception as e:
        print(e)

async def callback_test(mac):
    print("Connecting to mac: {0}".format(mac), flush=True)
    client = BleakClient(mac)
    client.set_disconnected_callback(__disconnect_callback)
    await client.connect()
    # do something with the connection, blah blah
    print("Poking services...".format(mac), flush=True)
    print(await client.get_services())
    # now disconnect
    print("Trying disconnect...".format(mac), flush=True)
    await client.disconnect()
    print("Disconnected. Awaiting disconnect callback.")
    await asyncio.sleep(20.0)

loop = asyncio.get_event_loop()
loop.run_until_complete(callback_test("24:71:89:cc:09:05"))

```

The disconnect callback is better used for handling when connection is lost than when you disconnect yourself. Windows is a bit slow with generating the callback, waiting until the connection is actually severed, which is NOT equal to Bleak's disconnect. I cannot do anything better than that I think, not with the API I use in Windows right now.

NbrTDB, 2020-09-25:
Fantastic, thanks very much! It seems giving asyncio something to do after requesting a disconnect allowed the event to trigger and I now get the "Disconnected from" output. It happened more or less immediately after trying to disconnect.

It seems I need to learn a bit more about how asyncio works; am I right in thinking that since the loop had "completed" after requesting a disconnect, there was no way the callback was ever going to fire?

In that case, do I need some way of keeping an event loop going the entire time I'm connected to a tool? My little bluetooth handler was intended to operate in response to commands from stdin (a command for "connect to this MAC address" and "read this characteristic" - that sort of thing). Each command is invoking `loop.run_until_complete()` separately, like shown below. Is this the wrong approach if I want callbacks to trigger? Once connected I can store the client instance and re-use it and it works nicely (callbacks notwithstanding), so I can connect to a tool, then some time later, send a command to read a gatt characteristic, etc.

```python
    def __stdin_msg_handler(self, inp):
        # read the input command and respond accordingly
        print("Received:", inp, flush=True)
        if(len(inp) > 0):
            # convert from JSON
            msg = json.loads(inp[0])
            if(msg["cmd"] == "a"):
                self.__hello()
            if(msg["cmd"] == "b"):
                loop = asyncio.get_event_loop()
                loop.run_until_complete(self.__find_devices())
            if(msg["cmd"] == "c"):
                mac = msg["mac"]
                loop = asyncio.get_event_loop()
                loop.run_until_complete(self.__connect_device(mac))                
            if(msg["cmd"] == "d"):
                mac = msg["mac"]
                loop = asyncio.get_event_loop()
                loop.run_until_complete(self.__disconnect_device(mac))                
            if(msg["cmd"] == "r"):
                mac = msg["mac"]
                service_uuid = "0000180a-0000-1000-8000-00805f9b34fb"
                characteristic_uuid = "00002a29-0000-1000-8000-00805f9b34fb"
                loop = asyncio.get_event_loop()
                loop.run_until_complete(self.__read_gatt_characteristic(mac, service_uuid, characteristic_uuid))                
            if(msg["cmd"] == "t"):
                mac = msg["mac"]
                loop = asyncio.get_event_loop()
                loop.run_until_complete(self.__callback_test(mac))
```

Also is there some way to *unbind* a disconnection callback? I note with the above approach the callback fires twice the second time, three times the third time... for now I figure I can just check if one is already bound and not bother doing so again for a given client.

hbldh, 2020-09-25:
Very good!

No, unbinding is not possible in Windows right now, due to several reasons. You should discard the client between uses and set up a new one each time to avoid it.

closed 2020-09-25 by hbldh

NbrTDB, 2020-09-25:
Thanks for all the advice, I really appreciate it!  

I've found a way around the need for `await asyncio.sleep()`; in short I just don't understand asyncio enough yet; the loop in my BLE handler that was listening for stdin was blocking...

I've changed things around such that the class has a singular event loop:

```python
def start(self):
        self.loop = asyncio.get_event_loop()
        self.loop.run_until_complete(self.__loop()) 
```

This makes the message handler (which is now async) much cleaner:

```python
    async def __stdin_msg_handler(self, inp):
        # read the input command and respond accordingly
        print("Received:", inp, flush=True)
        if(len(inp) > 0):
            # convert from JSON
            msg = json.loads(inp[0])
            if(msg["cmd"] == "a"):
                self.__hello()
            if(msg["cmd"] == "b"):
                await self.__find_devices()
            if(msg["cmd"] == "c"):
                # mac = msg["mac"]
                mac = "04:91:62:1C:D0:E0"
                await self.__connect_device(mac)         
            if(msg["cmd"] == "d"):
                # mac = msg["mac"]
                mac = "04:91:62:1C:D0:E0"
                await self.__disconnect_device(mac)         
           ...etc
```

And as for getting input from stdin:

```python
line = await self.loop.run_in_executor(None, sys.stdin.readline)
```

With these tweaks I don't need `asyncio.sleep()` and the disconnect callback fires as expected.