hbldh/bleak #309: Pair with devices that are not connected

hbldh, 2020-09-25:
Pairing was added in version 0.8.0. It still requires `connect` to be called first, due to that the OS native device communication objects being created in the `connect` methods (setting of the `self._requester` attributes). One could refactor this so that `pair` and `unpair` could be called on unconnected `BleakClient`s as well.

dlech, 2022-09-03:
Since pairing requires communicating with the remote device, I don't see how pairing could be done without connecting first.

dlech, 2022-09-20:
The BlueZ D-Bus API docs for "Pair" says:

> This method will connect to the remote device, initiate pairing and then retrieve all SDP records (or GATT primary services).

And the Windows API doesn't really have a notion of being "connected", so it seems like it should be possible to pair without an active gatt session. 

macOS doesn't allow explicit pairing, but rather prompts for it automatically the first time you try to access a characteristic that requires authentication.

The Android API doesn't say one way or the other. https://developer.android.com/reference/android/bluetooth/BluetoothDevice#createBond()

bojanpotocnik, 2022-11-25:
> And the Windows API doesn't really have a notion of being "connected", so it seems like it should be possible to pair without an active gatt session.

This is correct, tested with https://github.com/hbldh/bleak/pull/1133/commits/dd50aa66730f80e95ca43acd69b2e7a9f2e67043#diff-dca4aa1d374b221291b543a5d1296bbf912b43d1e2ebc51ed11c1eb153fbb89bR193-R196

PR #1741 (2025-05-10): add pair param to BleakClient()
This is an alternative to #1713 and fixes #309.
closed 2025-05-10 by dlech