hbldh/bleak #310: RemoteError when trying to disconnect

soldag, 2020-09-27:
* bleak version: 0.8.0
* Python version: 3.7.3
* Operating System: Raspbian Buster
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

When trying to close the connection of a device the following `RemoteError` is raised on my Raspberry Pi Zero. I can see in the logs of the other device that the connection was properly closed though, so the error seems to be raised afterwards.
Same code works well on macOS. 

### What I Did

```
client = BleakClient(address)
await client.connect()
await client.disconnect()

Traceback (most recent call last):
  File "/home/pi/venv/lib/python3.7/site-packages/aioconsole/execute.py", line 87, in aexec
    result, new_local = await coro
  File "<aexec>", line 2, in __corofn
  File "/home/pi/venv/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 245, in disconnect
    is_disconnected = not await self.is_connected()
  File "/home/pi/venv/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 346, in is_connected
    ).asFuture(asyncio.get_event_loop())
txdbus.error.RemoteError: org.freedesktop.DBus.Error.UnknownObject: Method "Get" with signature "ss" on interface "org.freedesktop.DBus.Properties" doesn't exist
```

dlech, 2020-09-28:
I haven't looked at the code yet, but I suspect we could fix this by catching this error in `is_connected` and return false if we get `org.freedesktop.DBus.Error.UnknownObject`.

soldag, 2020-10-04:
That's also what I did in my application code as a workaround so far and it worked fine. I created a PR for fixing this.

PR #317 (2020-10-06): Catch RemoteError in is_connected in BlueZ backend (#310)
Fixes #310
closed 2020-10-06 by dlech