hbldh/bleak #312: Disconnected callbacks firing multiple times when multiple connections are made in a session

NbrTDB, 2020-09-28:
* bleak version: 0.8.0
* Python version: 3.8.5
* Operating System: Windows 10 2004, build 19041.508

### Description

When connecting to a BLE device, I am creating a new client and setting a disconnected callback using `client.set_disconnected_callback()`, then storing the client in `self.connected_devices { }` with a reference against the MAC address.

This all works great including the firing off the callback when the device is connected (either by `client.disconnect()` or wandering out of range, powered off, etc). The disconnect callback discards the client by deleting it from `self.connected_devices` when fired.

However after reconnecting to the same device in the same session (and again creating a new client, assigning the disconnected callback and stuffing it in the dict of connected devices), on disconnect the callback with run twice. And on the third time, three times, and so on.

I have determined as a workaround, I could try separately tracking all MAC addresses (a `self.client_history` dict, perhaps) and then avoid setting disconnection callbacks on any MAC addresses I find in there, but I wonder if this is going about things the wrong way? Is there some way the client instance is surviving having been discarded? It's as if there is a stale reference to previously-connected devices somewhere.

### What I Did

```python
    async def __connect_device(self, mac):
        # connect to a device based on a mac address 
        print("Connecting to mac: {0}".format(mac),flush=True)
        client = BleakClient(mac)
        try:
            await client.connect()
        except BleakError as err:
            print(err)
        else:
            client.set_disconnected_callback(self.__disconnect_callback)
            await client.get_services()
            self.connected_devices[mac] = client

    async def __disconnect_device(self, mac):
        # connect to a device based on a mac address 
        print("Disconnecting mac: {0}".format(mac),flush=True)
        if mac in self.connected_devices:
            await self.connected_devices[mac].disconnect()

    def __disconnect_callback(self, device_client):
        # fires when a device is disconnected
        print("Disconnected from {0}".format(device_client.address))
        mac = device_client.address
        # remove from self.connected_devices
        if mac in self.connected_devices:
            del self.connected_devices[mac]
```

dlech, 2020-09-28:
Thanks for the feedback. The disconnect callback on Windows in new in v0.8.0, so we have probably overlooked something. I will look deeper when I have a chance.

NbrTDB, 2020-09-28:
No problem! On the subject of workarounds, maintaining a list of previously-connected MAC addresses and not setting callbacks on them again for the remainder of the session does seem to work quite nicely, insofar as even when connecting/disconnecting multiple different devices, a single disconnection event only ever triggered a single iteration of the disconnection callback.

dlech, 2020-09-28:
FYI, I have found a way to remove event handlers in pythonnet + winrt. Instead of using the `+=` operator, call the `add_EventHandlerName()` method. This returns a token that can be saved and later passed to `remove_EventHandlerName()`.

hbldh, 2020-09-28:
Amazing... that is how I do it in the `feature/winrt` branch. I have never seen that method in the `pythonnet` world; But it is a method on the EventBinding, right? I must have not `dir`:ed them enough. 

I prefer keeping solutions outside of the BleakBridge and if all event handler handling can be done with `add` and `remove` methods instead I am all for it!

dpelaez, 2020-10-07:
> No problem! On the subject of workarounds, maintaining a list of previously-connected MAC addresses and not setting callbacks on them again for the remainder of the session does seem to work quite [...]

I ran into the same problem and this workaround does work quite nicely.

commit 2f86b604 (2020-10-20): dotnet: Remove BluetoothLEDevice.ConnectionStatusChangedEvent handler on disconnect

When device was disconnected and reconnected, we would end up with
multiple disconnect handlers since the underlying .NET object was
reused.

By keeping the event registration token, we are able to remove the
event handler when we lose the connection to the device.

This should fix #312.
PR #340 (2020-10-20): Release/0.9.0
## Added

* Timeout for BlueZ backend connect call to avoid potential infinite hanging. Merged #306.
* Added Interfaces API docs again.
* Troubleshooting documentation.
* noqa flags added to ``BleakBridge`` imports.
* Adding a timeout on OSX so that the connect cannot hang forever. Merge #336.

## Changed

* ``BleakCharacteristic.description()`` on .NET now returns the same value as other platforms.
* Changed all adding and removal of .NET event handler from ``+=``/``-=`` syntax to calling ``add_`` and ``remove_`` methods instead. This allows for proper removal of event handlers  [... cut]
closed 2020-10-20 by hbldh