hbldh/bleak #314: disconnect_callback called on disconnect()

5frank, 2020-09-29:
* bleak version: 0.8.0
* Python version: 3.7
* Operating System: macOS Mojave

### Description

It seems that `bleak_client.disconnect()` have different behavior in `corebluetooth` and `bluezdbus` backend.

In corebluetooth backend, the user provided disconnect_callback will run even when the disconnect is intended - I assume this is not the desired behavior!?. In bluezdbus callback do not run (have not tested windows/dotnet).

I need the disconnect_callback to raise a error if a device unexpectedly disconnects, otherwise the application might hang forever.

I do currently not see a clean fix but maybe tomorrow ...

dlech, 2020-09-29:
From a user's point of view, I would expect a device disconnected callback to be called whenever a device was disconnected in all cases independent of the reason for the disconnection. This is how all other hardware libraries I have used behave, so this is the reason for my expectations.

5frank, 2020-09-30:
@dlech That makes sense I guess.  Regardless, most important thing is uniform behavior. A issue in bluezdbus then!?

Would be nice to with some mechanism to differentiate from a requested and a unexpected disconnect. I would appreciate if a BleakDisconnectException was raised on unexpected disconnect but that would probably break things.

hbldh, 2020-09-30:
I agree. A disconnected callback should be called on all types of disconnects. If you disconnect manually, then you can always remove the callback yourself prior to disconnecting to avoid it firing (`client.set_disconnected_callback(None)` would do it I think). Then you are free to raise an error of your choosing in the disconnected callback.

Bleak tries to offer a uniform API and uniform handling of output data, it can however never promise uniform behaviour. The different OS backends will have differences and it is regrettably up to the developer to learn and handle these for their own usecases. If BlueZ does not raise on regular disconnect, then that is regrettable but unavoidable right now. You will have to write code that accepts both behaviours then.

A disconnect/reconnect example will be written for #266, but I do not think I will raise an error in it. It will probably use `asyncio.Event` instead.

5frank, 2020-09-30:
@hbldh Thanks, will try that!

IIRC disconnect_callback was recently added to dotnet backend. What is the behavior? 
I do not have a working windows machine right now to test with.

dlech, 2020-10-01:
There are a few open issues regarding the disconnect behavior in the Windows backend currently. But the intention when I wrote it was to always call the disconnect callback.

hbldh, 2020-10-20:
I think this is now harmonized in version 0.9.0 just released. Will close for now.

closed 2020-10-20 by hbldh