hbldh/bleak #316: Data sent when Bleak connects to the device

A2va, 2020-10-01:
* bleak version: 0.8.0
* Python version: 3.8.5
* Operating System: Windows 10 Version 1903

### Description

I have a Nordic microcontroller that I programmed to logs the values received by Bluetooth according to the service. 

I have a Python script that connects to the microcontroller and when the connection is established, the microcontroller logs indicates that it has received a byte of value 2. While the program didn't send anything I also tried with an Android application to send the data but no log indicates any data received during the connection with the Android app

What should happen is that no data is exchanged during the connection.

### What I Did

The code has a txt extension because Github didn't want to upload a python file.

[bleak_send_data.txt](https://github.com/hbldh/bleak/files/5311252/bleak_send_data.txt)

dlech, 2020-10-01:
I have extracted the code from the file to make it easier for others to read.

```python
import asyncio
from bleak import BleakClient, discover
from aioconsole import ainput

CUSTOM_SERVICE_UUID="f3641400-00b0-4240-ba50-05ca45bf8abc"
CUSTOM_SERVICE_CHAR_UUID="f3641401-00b0-4240-ba50-05ca45bf8abc"

class Connection():
    
    client: BleakClient = None

    def __init__(self,loop:asyncio.AbstractEventLoop):
        self.connected = False
        self.connected_device = None

        self.loop = loop

    def on_disconnect(self, client: BleakClient, future: asyncio.Future):
        self.connected = False
        # Put code here to handle what happens on disconnet.
        print(f"Disconnected from {self.connected_device.name}!")
    
    async def cleanup(self):
        if self.client:
            await self.client.stop_notify(read_characteristic)
            await self.client.disconnect()

    async def manager(self):
        print("Starting connection manager.")
        while True:
            if self.client:
                await self.connect()
            else:
                await self.select_device()
                await asyncio.sleep(15.0, loop=loop)

    async def connect(self):
        if self.connected:
            return
        try:
            await self.client.connect()
            self.connected = await self.client.is_connected()
            if self.connected:
                print(F"Connected to {self.connected_device.name}")
                while True:
                    if not self.connected:
                        break
                    await asyncio.sleep(3.0, loop=loop)
            else:
                print(f"Failed to connect to {self.connected_device.name}")
        except Exception as e:
            print(e)

    async def select_device(self):
        print("Bluetooh LE hardware warming up...")
        await asyncio.sleep(2.0, loop=loop) # Wait for BLE to initialize.
        devices = await discover()

        print("Please select device: ")
        for i, device in enumerate(devices):
            print(f"{i}: {device.name}")

        response = -1
        while True:
            response = await ainput("Select device: ")
            try:
                response = int(response.strip())
            except:
                print("Please make valid selection.")
            
            if response > -1 and response < len(devices):
                break
            else:
                print("Please make valid selection.")

        print(f"Connecting to {devices[response].name}")
        self.connected_device = devices[response]
        self.client = BleakClient(devices[response].address, loop=self.loop)

async def main():
    while True:
        await asyncio.sleep(5)

async def user_console_manager(connection: Connection):
    while True:
        if connection.client and connection.connected:
            input_str = await ainput("Enter number: ")
            if input_str !='':
                write_value = bytearray([int(input_str)])
                await connection.client.write_gatt_char(CUSTOM_SERVICE_CHAR_UUID, write_value)
                print(f"Sent: {input_str}")
        else:
            await asyncio.sleep(2.0, loop=loop)

if __name__ == '__main__':
    loop = asyncio.get_event_loop()

    connection = Connection(loop)
    try:
        asyncio.ensure_future(connection.manager())
        asyncio.ensure_future(user_console_manager(connection))
        asyncio.ensure_future(main())
        
        loop.run_forever()
    except KeyboardInterrupt:
        print()
        print("User stopped program.")
    finally:
        print("Disconnecting...")
        loop.run_until_complete(connection.cleanup())
```

A2va, 2020-10-02:
Thanks for suggested modification of code. It there be a fix for this ? Otherwise I can modify the code of microcontroller to ignore the first byte.

hbldh, 2020-10-06:
I do not think there can be a fix, since I am unsure of where the extra byte is originating from. I doubt that it is in the Bleak code-base, but perhaps in the OS BLE API code.

dlech, 2020-10-20:
You could try logging Bluetooth traffic to see where the extra byte is coming from: https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10

A2va, 2020-10-27:
Ok. Thanks !

dlech, 2022-07-25:
closing due to inactivity

closed 2022-07-25 by dlech