hbldh/bleak #319: 'NoneType' object has no attribute 'delegate' exception when connecting

0ge, 2020-10-05:
* bleak version: 0.8.0
* Python version: Python 3.7.4
* Operating System: macOS 10.15.2 (Catalina)

### Description

I'm trying to write an application that uses tkinter and bleak to discover and interact with a custom BLE device.
However, when I try to connect to the device, it throws an exception:

```
'NoneType' object has no attribute 'delegate'
  File ".../scratch.py", line 29, in run
    await client.connect()
  File ".../miniconda3/envs/py3/lib/python3.8/site-packages/bleak/backends/corebluetooth/client.py", line 86, in connect
    manager = self._device_info.manager().delegate()
```

(There is also a similar exception thrown in the final clause when attempting to disconnect)

I have condensed this to the application shown below.

```
import asyncio
import re
import traceback
import tkinter as tk
from bleak import BleakClient
from bleak import BleakScanner

async def run():
    if explicit:
        devices = await BleakScanner.discover(loop=loop)
    else:
        devices = await BleakScanner.discover()

    for d in devices:
        print(d)

    matcher = re.compile(r"[A-z]{3}-[0-9]{3}")
    devices = [d for d in devices if matcher.match(d.name)]
    if explicit:
        client = BleakClient(devices[0].address, loop=loop)
    else:
        client = BleakClient(devices[0].address)

    try:
        if explicit:
            await client.connect(loop=loop)
        else:
            await client.connect()
        print("Connected to {0}".format(client))
    except Exception as e:
        print(e)
        traceback.print_tb(e.__traceback__)
    finally:
        await client.disconnect()

def scan_and_connect():
    asyncio.get_event_loop().create_task(run())

async def update_ui():
    while True:
        root.update()
        await asyncio.sleep(1 / 30)

explicit = False  # Explicit or implicit run loops
loop = asyncio.get_event_loop()

root = tk.Tk()
if explicit:
    root.loop = loop
btn = tk.Button(root, text="Scan and connect", command=scan_and_connect)
btn.grid(column=1, row=1)

loop.run_until_complete(update_ui())

# Works as expected (i.e. connects to device)
# loop.run_until_complete(run())
```

### What I Did

I've tried the following combinations:

Bleak | Loop      | Outcome
------|---------|-----------
v0.7.1 | Implicit  | Exception
v0.7.1 | Explicit  | Exception
v0.8.0 | Implicit | Exception
v0.8.0 | Explicit | Exception

dlech, 2020-10-05:
Since Bleak 0.8.0, it is possible to do this:

    BleakClient(devices[0])

Instead of this:

    BleakClient(devices[0].address)

Passing the address causes the device to get discovered twice, which can cause problems for some devices. Also the `loop` keyword argument is removed in v0.8.0, so it is ignored.

0ge, 2020-10-05:
Sorry, `.address` was left from 0.7.1 testing. Same results though. Using explicit loops was just to rule out that it was related to the deprecation.

I hope to be able to test this on a Windows machine sometime soon, but can't test at the moment.

Carglglz, 2020-10-06:
See #264 
The reason is `self._device_info` is a `CBPeripheral` object and calling the method `self._device_info.manager()` should return a `CBCentralManager` object which has a `delegate` attribute indeed.

In this case `self._device_info.manager()` returns `None` instead and error `AttributeError: 'NoneType' object has no attribute 'delegate'` is raised.

I'm not sure about why this happens, but I suspect it may be a "_discovering_ / _timing_" issue as previously said.

@dlech 
I think there is actually a "bug"  in `bleak/backends/corebluetooth/client.py` which causes
```
BleakClient(devices[0])
```
To be ignored and still try to discover the device with `BleakScannerCoreBluetooth`

I think the last `self._device_info = None` is left over ?
```
  def __init__(self, address_or_ble_device: Union[BLEDevice, str], **kwargs):
        super(BleakClientCoreBluetooth, self).__init__(address_or_ble_device, **kwargs)

        if isinstance(address_or_ble_device, BLEDevice):
            self._device_info = address_or_ble_device.details
        else:
            self._device_info = None

        self._device_info = None
        self._requester = None
        self._callbacks = {}
        self._services = None
```

dlech, 2020-10-06:
> I think the last `self._device_info = None` is left over ?

Yes, I agree that should not be there.

Carglglz, 2020-10-06:
Doing a PR now! 👍

0ge, 2020-10-07:
I tried the PR and it gives a more informative message now. It still fails to connect, though.

```
2020-10-07 10:32:51,220 - bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG - Discovered device E474B4BE-9F13-4036-A1A8-A67D53735BE1: MyDevice @ RSSI: -75 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>)
Device with address E474B4BE-9F13-4036-A1A8-A67D53735BE1 was not found
  File "/Users/user/scratch.py", line 22, in run
    await client.connect()
  File "/Users/user/opt/miniconda3/envs/py3/lib/python3.8/site-packages/bleak/backends/corebluetooth/client.py", line 90, in connect
    raise BleakError(
```

Carglglz, 2020-10-07:
Did you try with `BleakClient(devices[0])` method?

Could you try this:
```
import asyncio
from bleak import discover

def ble_scan(log=False):
    devs = []

    async def run():
        devices = await discover()
        for d in devices:
            if log:
                print(d)
            devs.append(d)
        return devs

    loop = asyncio.get_event_loop()
    return loop.run_until_complete(run())

>>> devs = ble_scan()
>>> devs
[B1EDD56B-AEE4-4E1A-A235-376FCED5E242: LG6, E0F4D80F-0E74-416D-91FF-986E012CB97A: Unknown]
>>> devs[0].details
<CBPeripheral: 0x7fc7fe02c0a0, identifier = B1EDD56B-AEE4-4E1A-A235-376FCED5E242, name = LG6, state = disconnected>
>>> devs[0].details.manager()
<CBCentralManager: 0x7fc7fe010240>
>>> devs[0].details.manager().delegate()
<CentralManagerDelegate: 0x7fc7fe008aa0>

```

This `devs[0].details.manager()` should return a `CBCentralManager`.

If this does not work:

Have you tried any of the scripts in `bleak/examples` ,  or with another device (phone) ?

Could you try to place the device as closest as possible and try again? (at  RSSI (-60, -50) at least)

What's the advertising interval of the device? (the smaller the easier to discover and connect)

0ge, 2020-10-07:
Yes, only using BLEDevice returned from scanning now. I also added som logging to bleak and found that the manager is `None` when the client is instantiated, if that is of any help.

I will try your code tomorrow and check advertising interval as well. All examples work. As mentioned, it works if I remove tkinter from the equation and it works_with_ tkinter on Windows. Don't have any Android access, sorry.

I can connect with CoreBluetooth in both iOS and macOS applications without any problems, so I don't think the issue is with CB here. (It also works with other apps on both iOS, Windows and a niche OS (smartBASIC))

0ge, 2020-10-08:
I added logging to try to be of help and found something peculiar. Here is the patch I applied to bleak:
```
diff --git a/bleak/backends/corebluetooth/client.py b/bleak/backends/corebluetooth/client.py
index c7456a0..60b2d87 100644
--- a/bleak/backends/corebluetooth/client.py
+++ b/bleak/backends/corebluetooth/client.py
@@ -50,6 +50,9 @@ class BleakClientCoreBluetooth(BaseBleakClient):
         else:
             self._device_info = None
 
+        logger.debug(self._device_info)
+        logger.debug(self._device_info.manager())
+
         self._requester = None
         self._callbacks = {}
         self._services = None
diff --git a/bleak/backends/corebluetooth/scanner.py b/bleak/backends/corebluetooth/scanner.py
index 9faf384..c66c07c 100644
--- a/bleak/backends/corebluetooth/scanner.py
+++ b/bleak/backends/corebluetooth/scanner.py
@@ -106,6 +106,11 @@ class BleakScannerCoreBluetooth(BaseBleakScanner):
                 for u in advertisementData.get("kCBAdvDataServiceUUIDs", [])
             ]
 
+            logger.debug("BLEDevice details: {0}".format(details))
+            logger.debug("BLEDevice manager: {0}".format(details.manager()))
+            logger.debug(
+                "BLEDevice manager delegate: {0}".format(details.manager().delegate())
+            )
             found.append(
                 BLEDevice(
                     address,
```

Suddenly, it succeeds in connecting. However, it fails similarly when trying to disconnect:

```
2020-10-08 09:01:31,937 - bleak.backends.corebluetooth.scanner - DEBUG - BLEDevice details: <CBPeripheral: 0x7fbb8195a880, identifier = E3789FBC-2BAA-4175-9E40-6DDEF8BEA332, name = MyDevice, state = disconnected>
2020-10-08 09:01:31,938 - bleak.backends.corebluetooth.scanner - DEBUG - BLEDevice manager: <CBCentralManager: 0x7fbb818d7780>
2020-10-08 09:01:31,939 - bleak.backends.corebluetooth.scanner - DEBUG - BLEDevice manager delegate: <CentralManagerDelegate: 0x7fbb818d9580>
2020-10-08 09:01:32,050 - bleak.backends.corebluetooth.client - DEBUG - <CBPeripheral: 0x7fbb8195a880, identifier = E3789FBC-2BAA-4175-9E40-6DDEF8BEA332, name = MyDevice, state = disconnected>
2020-10-08 09:01:32,051 - bleak.backends.corebluetooth.client - DEBUG - <CBCentralManager: 0x7fbb818d7780>
2020-10-08 09:01:32,051 - bleak.backends.corebluetooth.client - DEBUG - Connecting to BLE device @ E3789FBC-2BAA-4175-9E40-6DDEF8BEA332
2020-10-08 09:01:32,471 - bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG - centralManager_didConnectPeripheral_
2020-10-08 09:01:32,472 - bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG - Successfully connected to device uuid E3789FBC-2BAA-4175-9E40-6DDEF8BEA332
2020-10-08 09:01:32,474 - bleak.backends.corebluetooth.client - DEBUG - Retrieving services...
2020-10-08 09:01:32,475 - bleak.backends.corebluetooth.PeripheralDelegate - DEBUG - peripheral_didDiscoverServices_
2020-10-08 09:01:32,475 - bleak.backends.corebluetooth.PeripheralDelegate - DEBUG - Services discovered
2020-10-08 09:01:32,475 - bleak.backends.corebluetooth.client - DEBUG - Retrieving characteristics for service C8D5036B-806E-3DB0-DC4D-335CFF5A6DDA
...
2020-10-08 09:01:33,215 - bleak.backends.corebluetooth.client - DEBUG - Services resolved for BleakClientCoreBluetooth (E3789FBC-2BAA-4175-9E40-6DDEF8BEA332)
Connected!
Traceback (most recent call last):
  File "/Users/user/scratch.py", line 28, in run
    await client.disconnect()
  File "/Users/user/bleak/bleak/backends/corebluetooth/client.py", line 114, in disconnect
    manager = self._device_info.manager().delegate()
AttributeError: 'NoneType' object has no attribute 'delegate'
```

Carglglz, 2020-10-08:
>  All examples work. As mentioned, it works if I remove tkinter from the equation

I think this is  the key, and I was about to suggest adding `CBPeripheral`, `CBCentralManager` and `CentralManagerDelegate` to logging exactly as you did 👍 .

> Suddenly, it succeeds in connecting. However, it fails similarly when trying to disconnect:

This makes me think it could be a memory allocation problem, while using tkinter or for some reason `CBPeripheral` losing reference of the `CBCentralManager` so calling `.manager()` returns `None`. If that's the case I will try to pass
`CBCentralManager` or even `CentralManagerDelegate`  directly to BLEDevice so there will no need to call
`self._device_info.manager().delegate()`.

I just push a new branch `fix-corebluetooth-delegate` in my fork with these changes. 
See if that makes any difference 👀

0ge, 2020-10-08:
Works perfectly! 👌
Both connecting and disconnecting works with tkinter now. I'll continue testing next week with reading, writing and notifications.

dlech, 2020-10-09:
Thanks @0ge and @Carglglz for all of the investigation and testing to get to the bottom of the issue!

0ge, 2020-10-16:
Sorry that it took an awful long time, but I have now confirmed that reading, notifications and writings works as expected. From my side, this has been resolved.

closed 2020-10-16 by 0ge