hbldh/bleak #320: A way to detect BT module not being present or disabled at all

SteKme, 2020-10-05:
* bleak version: 0.8.0
* Python version: 3.7x64
* Operating System: Windows 10

### Description
Discovery service - BleakScanner.discover() - returns [] no matter if 
** there are no devices the PC can connect to (which is fine, indeed), 
** but also when the BT module of the PC is turned off (my case, notebook's BT module off in the right hand command control center of Windows)
** ? or the BT module not present at all (can't check if this applies as well at the moment).

I can't seem to find any way to detect the BT module of the PC not being present / disabled in the Windows. Would it be perhaps possible to throw an exception or something in this discover() when there is no BTLE hardware connected to the PC at all? Any other working solution is fine, of course.

Thank You so much!

hbldh, 2020-10-06:
Currently, Bleak has no capability to check if your computer's Bluetooth adapter is on or off. There are API methods that can check it, and in one case it seems it can also switch them on and off (BlueZ).

- BlueZ: https://kernel.googlesource.com/pub/scm/bluetooth/bluez/+/refs/tags/5.52/doc/adapter-api.txt#243
- Windows: https://stackoverflow.com/a/48794678 and https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.bluetoothadapter.getradioasync?view=winrt-19041#Windows_Devices_Bluetooth_BluetoothAdapter_GetRadioAsync
- Core Bluetooth: This might be used as a hint: https://stackoverflow.com/a/57953922

It is not a priority task for me I am afraid. You are more than welcome to do the implementation and make a PR, because I approve of the functionality being present.

dlech, 2020-10-22:
We are actually already monitoring this in CoreBluetooth:

https://github.com/hbldh/bleak/blob/9ff5ed8ccbb0493e2994a0bbfe9ebf26810878fb/bleak/backends/corebluetooth/CentralManagerDelegate.py#L178-L195

But the state is not saved anywhere.

We should also revisit the use of `wait_for_powered_on()` in CoreBluetooth when we look at this. According to the CoreBluetooth docs, we should be waiting for this to be resolved before calling any other APIs, but I don't think we are doing that in all cases. (Note to self: we can probably use a `threading.Event` instead of `asyncio.Event` and make the `__init__()` method blocking when we create a new central manager - it shouldn't block for long.)

0ge, 2022-12-05:
We have ran into this for a tool we developed. One user frequently turned off his Bluetooth radio (because of other reasons) but the tool has no way of informing him that the radio is turned off. For our use case, it would work with either an exception when attempting to scan, but preferably a separate method/property/function that we could check so that the user can be informed before attempting to scan.

Our current workaround is to display a reminder to check if the radio is off when no devices are discovered.