hbldh/bleak #321: GATT Server connected & can write but cannot read services

Arunprasad657, 2020-10-06:
* bleak version:0.8.0
* Python version:3.7
* Operating System:Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: 

### Description
I want to send  command to device after that I want to receive that response of the particular command. I'm able to see the response in output log screen but I can't read via using read_gatt_char.

### What I Did
import asyncio
from bleak import BleakClient
import time
from bleak import discover
import platform
import logging
import asyncio

from bleak import BleakClient
from bleak import _logger as logger
from bleak.uuids import uuid16_dict

address = "A4:CF:18:8B:A9:DE"
MODEL_NBR_UUID = "00002a00-0000-1000-8000-00805f9b34fb"
SERVICE_UUID = "0000a0002-0000-1000-8000-00805f9b34fb"
IO_DATA_CHAR_UUID_W = "0000c303-0000-1000-8000-00805f9b34fb"
IO_DATA_CHAR_UUID_R = "0000c306-0000-1000-8000-00805f9b34fb"

def keypress_handler(sender, data):
    print("{0}: {1}".format(sender, data))

def callback(sender: int, data: bytearray):
    print(f"{sender}: {data}")

async def run(address, loop):
    read_characteristic: str
    write_characteristic: str
    devices = await discover()
    for d in devices:
        print(d)

    async with BleakClient(address, timeout=1.0,  loop=loop) as client:
        #client.write_gatt_char("Welcome");
        x = await client.is_connected()
        print("Connected")
        if (x == True):
            service = await client.get_service(SERVICE_UUID)
            #client.start_notify(IO_DATA_CHAR_UUID_R, callback)
            #print("Services : ".format( await client.get_services()))
            model_number = await client.read_gatt_char(MODEL_NBR_UUID)
            print("Model Number: {0}".format("".join(map(chr, model_number))))

            #await client.start_notify(IO_DATA_CHAR_UUID_R, keypress_handler)

            write_value = bytearray(b'%batterychgst=0x0#')

            await client.write_gatt_char(IO_DATA_CHAR_UUID_W, write_value)
            print("I/O Data Writed Value: {0}".format(write_value))

            await asyncio.sleep(3.0, loop=loop) # wait for a response
            print("Done!")

            value = await client.read_gatt_char(IO_DATA_CHAR_UUID_R)
            print("I/O Data Read Value: {0}".format(value))

            #await asyncio.sleep(5.0, loop=loop)
            #await client.stop_notify(IO_DATA_CHAR_UUID_R)
        else:
            print("Not Connected")

loop = asyncio.get_event_loop()
loop.run_until_complete(run(address, loop))

In log terminal I can connect and Write a data but not able to read the Writed data value Response

hbldh, 2020-10-06:
> In log terminal I can connect and Write a data but not able to read the Writed data value Response

What do you mean with not able to read? Does it throw an exception? Is it the wrong value?

Arunprasad657, 2020-10-06:
wrong value

hbldh, 2020-10-06:
Can you use any other BLE package (on phone or other desktop software) to write the same value and get the correct value read?

Arunprasad657, 2020-10-07:
yes, I used same UUID for READ and Write in Android App ie (IO_DATA_CHAR_UUID_W = "0000c303-0000-1000-8000-00805f9b34fb"
IO_DATA_CHAR_UUID_R = "0000c306-0000-1000-8000-00805f9b34fb")it's works fine on android app

hbldh, 2020-10-09:
Well, then I do not know what to do with this. I cannot reproduce it.

Try debugging on your peripheral side and see what it actually receives. Can you send something simpler or send to any other characteristic to see if it register writes at all?

dlech, 2022-07-25:
closing due to inactivity

closed 2022-07-25 by dlech