hbldh/bleak #322: [Question] How to request while getting answer with multiple clients

sneko, 2020-10-06:
* bleak version: v0.8.0
* Python version: 3.5+
* Operating System: Linux/MacOS

## Description

Hi,

I'm really new in the BLE area but read a bunch of forums about how to do the right things but things are still unclear and it seems it could depend on the library but also the underlying BLE device used (which seems logic :) ).

My context:
- I have a central entity (A) that needs to communicate with devices around (let's say they are type (B))
- (A) should communicate with all (B) devices at different timing (maybe almost at the same time)

After searching a few days it seems there is no trivial way to do some "request/response" in a synchronous way like we could do with HTTP call or anything else. I found 2 ways:

### First

1. `await write_gatt_char(...)`
2. `await sleep(...)` (the time to wait depends on how long the command should take at maximum on the (B) devices)
3. `await read_gatt_char(...)`

What I don't like with this one is I need to guess a time to wait before reading the answer.

### Second

Source: from @hbldh https://github.com/hbldh/bleak/issues/59#issuecomment-482503333

```
await client.start_notify(rw_charac, notification_handler)
await client.write_gatt_char(rw_charac, b'~HELLO', response=True)
await asyncio.sleep(0.5, loop=loop)  # Sleeping just to make sure the response is not missed...
await client.stop_notify(rw_charac)
```

In this example, contrary to the first one, the callback will be triggered as soon as the device has sent it. So we do not wait for reading, it's immediate.

The example just shows uses a `sleep()` to make sure we handle notifications for at least the right time. I guess that if I have a long-running BLE client, I could not stop handling notifications (except when terminating the process), so the `sleep()` would be useless:
```
await client.write_gatt_char(rw_charac, b'~HELLO', response=True)
```

### Question

What way do you recommend? Maybe a third one exists?

Also, since (A) should reach multiple (B) devices, is it possible with `bleak` to maintain all the BLE clients connected? Does the library manage it correctly under the hood (no conflict...)? Because my "asyncio mono-thread" during an `await` could run another write/read operation on another device. So should I use a mutex/lock wrapping all operations like this:
```
common_mutex.lock()
ble_clients[x].connect()
ble_clients[x].write() (and so on)
ble_clients[x].disconnect()
common_mutex.unlock()
```

Sorry for all the text, really glad start using BLE and `bleak` 👍 

Thank you,

hbldh, 2020-10-09:
First of all, you should not use any mutex. Or at least, I do not think that you need them and that you are better off without them in this case.

But yes, connecting simultaneously to multiple clients is definitely possible: e.g. take a look at the [two_devices.py](https://github.com/hbldh/bleak/blob/develop/examples/two_devices.py) example. My recommended way is to keep clients separate in different coroutines, but there are other ways to do it as well; look in previous (closed) issues regarding multiple clients.

> the sleep() would be useless

In `asyncio`, `sleep` is never useless. It signals that you have nothing active to do right now and releases control back to the event loop, allowing it to process other things. It allows for handling the notification callbacks for instance.

sneko, 2020-10-09:
> First of all, you should not use any mutex. Or at least, I do not think that you need them and that you are better off without them in this case.

Maybe I didn't understand well how the `connnect/disconnect` work?

From the bluetooth documentation it seems the theorical number of "clients" is 7 (8 with the one communicating). But in practice with BLE people seems to say 3-4 sounds like the maximum.

So if I don't put any locker/mutex to connect/communicate/disconnect, what would happen? How does Bleak manage it if I try to have more than 7 remote connected devices?

> In asyncio, sleep is never useless. It signals that you have nothing active to do right now and releases control back to the event loop, allowing it to process other things. It allows for handling the notification callbacks for instance.

Indeed, I was just refering to the specific case when the connection to each device would be long-living (hours or so), so the `stop_notifying` could be used when `disconnecting`, not just after sending a command while waiting for it to stop the notifications.

Thank you,

Note: will dig in previous issues even if I probably saw them ^^

hbldh, 2020-10-09:
I have never connected to more than two simultaneously with Bleak, but theoretically it should work with more. There are some different limitations in different OS:es, but I am not sure the exact numbers or recommendations.

> So if I don't put any locker/mutex to connect/communicate/disconnect, what would happen? How does Bleak manage it if I try to have more than 7 remote connected devices?

It registers callbacks on separate connection objects received from the OS BLE APIs. All connected devices are kept separate by the OS and Bleak utilizes that separation. If you are really interested in how it is done, read the source code. 

What happens if you have to many? No idea. Find out and document it here!

closed 2021-03-03 by dlech