hbldh/bleak #331: macOS connection erorr: "objc.error: NSInternalInconsistencyException - Invalid parameter not satisfying: [identifiers isKindOfClass:[NSArray class]]"

koreymacdougall, 2020-10-13:
* bleak version: 0.8.0
* Python version:  3.8.5
* Operating System: macOS 10.11.6
* BlueZ version (`bluetoothctl -v`) in case of Linux: n/a

### Description

First off, great work on this library @hbldh, @dlech, et al.!

I am unable to connect to my BTLE device.  The same code works fine on Linux (Debian 10).  Program crashes upon "client.connect()", and throws an error referencing Invalid parameters.  I'm not sure if this issue is at the level of the OS, objective-c, or Bleak.  Any help would be greatly appreciated.  Code (simple sample from discussions in this repo) and stack trace below.

### What I Did

#### Code:
```
import asyncio
from bleak import BleakClient, discover

async def run():
    devices = await discover()
    for d in devices:
            print(d)
            client = BleakClient(d.address)
            await client.connect()

loop = asyncio.get_event_loop()
loop.run_until_complete(run())

```

#### Stack trace:
```
2020-10-12 22:16:31.801 Python[3820:332546] *** Assertion failure in -[CBCentralManager retrievePeripheralsWithIdentifiers:], /BuildRoot/Library/Caches/com.apple.xbs/Sources/CoreBluetooth/CoreBluetooth-102.23/CBCentralManager.m:203
Traceback (most recent call last):
  File "bleak_test3.py", line 52, in <module>
    loop.run_until_complete(run())
  File "/usr/local/Cellar/python@3.8/3.8.5/Frameworks/Python.framework/Versions/3.8/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "bleak_test3.py", line 12, in run
    await client.connect()
  File "/Users/GirlOne/dev/formid_hfs/venv/lib/python3.8/site-packages/bleak/backends/corebluetooth/client.py", line 76, in connect
    device = await BleakScannerCoreBluetooth.find_device_by_address(
  File "/Users/GirlOne/dev/formid_hfs/venv/lib/python3.8/site-packages/bleak/backends/corebluetooth/scanner.py", line 157, in find_device_by_address
    return await scanner._find_device_by_address(
  File "/Users/GirlOne/dev/formid_hfs/venv/lib/python3.8/site-packages/bleak/backends/scanner.py", line 107, in _find_device_by_address
    for d in await self.get_discovered_devices()
  File "/Users/GirlOne/dev/formid_hfs/venv/lib/python3.8/site-packages/bleak/backends/corebluetooth/scanner.py", line 82, in get_discovered_devices
    peripherals = self._manager.central_manager.retrievePeripheralsWithIdentifiers_(
objc.error: NSInternalInconsistencyException - Invalid parameter not satisfying: [identifiers isKindOfClass:[NSArray class]]

```

dlech, 2020-10-13:
Hmm... I guess old version of macOS (10.11) or PyObjC doesn't know how to convert a Python array to an NSArray in 

```python
        peripherals = self._manager.central_manager.retrievePeripheralsWithIdentifiers_(
            self._identifiers.keys(),
        )
```

Does changing it like this do anything?

```python
from Foundation import NSArray

...

            NSArray(self._identifiers.keys()),
```

koreymacdougall, 2020-10-13:
That did the trick!  Wow.  Thanks so much for the quick response!

dlech, 2020-10-13:
Cool, thanks for trying it out.

koreymacdougall, 2020-10-13:
Closing, as issue resolved.

Also note: upgrading the offending machine to macOS 10.15 Catalina solved the problem as well.

PR #380 (2020-12-07): Fixed ``BleakScanner.discover()`` on older versions of macOS.
Fixes #331
Fixes #352
commit 4591364c (2020-12-07): Fixed ``BleakScanner.discover()`` on older versions of macOS.

Fixes #331
Fixes #352
PR #391 (2020-12-11): Release/v0.10.0
## Added

* Added ``AdvertisementData`` class used with detection callbacks across all
  supported platforms. Merged #334.
* Added ``BleakError`` raised during import on unsupported platforms.
* Added ``rssi`` parameter to ``BLEDevice`` constructor.
* Added ``detection_callback`` kwarg to ``BleakScanner`` constructor.

## Changed

* Updated minimum PyObjC version to 7.0.1.
* Consolidated implementation of ``BleakScanner.register_detection_callback()``.
  All platforms now take callback with ``BLEDevice`` and ``AdvertisementData``
  arguments.
* Consolidated ``BleakScanner.find_ [... cut]
closed 2020-10-13 by koreymacdougall