hbldh/bleak #332: BLE problems on Raspberry Pi 4

dhalbert, 2020-10-13:
* bleak version: 0.8.0
* Python version: 3.7.3
* Operating System: Raspberry Pi OS buster (completely up to date) **on Raspberry Pi 4**
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

This is not a bleak problem _per se_, I believe.

I'm trying to get bleak to work on an RPi 4. It works fine on an RPi Zero W, but swapping the SD card from the RPi Zero W into the RPi 4 (assuring that the OS setup is identical) gives this error when trying to do service discovery:

```
...
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 154, in connect
    await self.get_services()
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 384, in get_services
    raise BleakError("Services discovery error")
```
The `properties` value `ServicesResolved` never becomes `True`, which causes the error.

If anyone has gotten bleak to work on an RPi4, I'd love to hear what you had to do. Thanks.

dhalbert, 2020-10-13:
This appears to be a `bluez-firmware` problem which manifests on more recent RPi's. Discussion here: https://github.com/adafruit/Adafruit_CircuitPython_BLE_BerryMed_Pulse_Oximeter/issues/3#issuecomment-708007815

> This appears to be a bluez-firmware problem. As of this writing, the latest version is bluez-firmware-1.2-4+rpt5. I had to roll back to bluez-firmware-1.2-4+rpt2 to get this to work. I downloaded the .deb from http://archive.raspberrypi.org/debian/pool/main/b/bluez-firmware/bluez-firmware_1.2-4+rpt2_all.deb. Versions rpt3 and rpt4 didn't work. What seems like the same problem was reported here: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=285641&p=1728626, and that report rolled back to rpt2 also.

hbldh, 2020-10-20:
This is a highly interesting find; thank you for documenting it! I will add a link to this issue on the new `troubleshooting` page in the docs (#330).

dhalbert, 2020-11-20:
Reopening this for your interest because there is new firmware under test at https://github.com/RPi-Distro/bluez-firmware/issues/6. Unfortunately it does not work for me.

dhalbert, 2020-11-20:
> Reopening this for your interest because there is new firmware under test at [RPi-Distro/bluez-firmware#6](https://github.com/RPi-Distro/bluez-firmware/issues/6). ~~Unfortunately it does not work for me.~~

Turns out for some reason I had to power cycle the board (not just reboot) in order to test the new firmware properly. It now does work for me, getting values from a pulse oximeter via our software on top of bleak.

dhalbert, 2021-03-21:
The `bluez-firmware-1.2-4+rpt7` and `rpt8` versions fix the problem mentioned. Use `rpt8`, which includes security fix not in `rpt7`.

closed 2021-03-21 by dhalbert

andysfigfi, 2022-04-06:
I'm getting this error on my RaspbPi 3+ B   
Any ideas what could be worng?

```PuTTYSPPService::read
Retries left, tries  2
BoundWriteStream::write B
Traceback (most recent call last):
  File "code.py", line 441, in <module>
    run()
  File "code.py", line 389, in run
    spp_service.write(pkt.build())
  File "code.py", line 193, in write
    self._tx.write(buf)
  File "code.py", line 47, in write
    self.bound_characteristic.value = buf[offset:offset + (mylen-offset) ]
  File "/home/pi/.local/lib/python3.7/site-packages/_bleio/common.py", line 579, in value
    response=self.properties & Characteristic.WRITE,
  File "/home/pi/.local/lib/python3.7/site-packages/_bleio/common.py", line 140, in await_bleak
    return future.result(timeout)
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 432, in result
    return self.__get_result()
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 812, in write_gatt_char
    assert_reply(reply)
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name)
bleak.exc.BleakDBusError: org.bluez.Error.NotPermitted
```