hbldh/bleak #335: Bluez: Exception thrown when properties changed callback called

lchish, 2020-10-13:
* bleak version: 0.8.0 git 99f3b7
* Python version: 3.7
* Operating System: raspbian
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.

### What I Did
Happens the second time I try to connect to and communicate with a bluetooth device.

I recently updated to 0.8.0 and modified my connection code to use the BleakDevice instead of address as it connects much faster.
 
The code is too long and complicated to paste into this bug report. At a high level it turns on a device using NFC and scans for a bluetooth device that matches an expected name:
                devices = await scanner.get_discovered_devices()
                for d in devices:
                    if self.ble_id in d.name:
                        self.bt_device = d
Then it connects to that device and sets a notification callback:
        self.client = bleak.BleakClient(self.bt_device)
        await self.client.connect()
        await self.client.start_notify(UART_TX, self._notification_callback)

Commands are then sent to it:
    async def _bt_send_cmd(self, command):
            await self.client.write_gatt_char(UART_RX, bytearray(command.encode()))

And this is where the exception is thrown happens.

I did a bit of debugging and self.address in bluezdbus/client.py is of type BleakClient instead of being a string. Perhaps _properties_changed_callback was never updated for the new way of sending devices to the client constructor?

Here's the exception I'm receiving:
Unhandled Error
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/txdbus/protocol.py", line 140, in dataReceived
    self.rawDBusMessageReceived(raw_msg)
  File "/home/pi/.local/lib/python3.7/site-packages/txdbus/protocol.py", line 266, in rawDBusMessageReceived
    self.signalReceived(m)
  File "/home/pi/.local/lib/python3.7/site-packages/txdbus/client.py", line 642, in signalReceived
    self.router.routeMessage(msig)
  File "/home/pi/.local/lib/python3.7/site-packages/txdbus/router.py", line 128, in routeMessage
    r.match(m)
--- <exception caught here> ---
  File "/home/pi/.local/lib/python3.7/site-packages/txdbus/router.py", line 67, in match
    self.callback(m)
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 866, in _properties_changed_callback
    self.address.replace(":", "_"),
builtins.AttributeError: 'BLEDevice' object has no attribute 'replace'

 Traceback (most recent call last):
   File "ma.py", line 589, in <module>
     mt = test.run_test("test_serial")
   File "ma.py", line 521, in run_test
     if not mt.test_comms():
   File "ma.py", line 323, in test_comms
     self.bt_send_cmd('$')
   File "monitor_assembly.py", line 242, in bt_send_cmd
     self.loop.run_until_complete(self._bt_send_cmd(command, sleep))
   File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
     return future.result()
   File "a.py", line 247, in _bt_send_cmd
     await self.client.write_gatt_char(UART_RX, bytearray(command.encode()))
   File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 629, in write_gatt_char
     ).asFuture(asyncio.get_event_loop())
 txdbus.error.RemoteError: org.bluez.Error.Failed: Not connected

NonsenseInc, 2020-10-20:
As a workaround you can change line 866 of the BlueZ client.py (https://github.com/hbldh/bleak/blob/29866b56ebf8563c7bae5ae339781bf25ae78c54/bleak/backends/bluezdbus/client.py#L866) to `str(self.address).replace(":", "_"),`. The error seems to appear, because it expects an address string, but `self.address` can also be an object.
I don't think this is a proper fix, so I'm not making a patch/pull-request.

hbldh, 2020-10-20:
This was a bug introduced in version 0.8.0, which I believe is solved by #311. The fix will be present in the next release.

closed 2020-10-20 by hbldh