hbldh/bleak #337: Event loop not running in CoreBluetooth PeripheralDelegate

adamincera, 2020-10-15:
* bleak version: 0.8.0
* Python version: 3.8.5
* Operating System: macOS 10.15.7
* BlueZ version (`bluetoothctl -v`) in case of Linux: N/A

### Description

I enabled notifications on a characteristic on a connected peripheral, but my callback was never executing. Upon further investigation, I determined that `peripheral_didWriteValueForCharacteristic_error_` is being called, and the script has received the expected data from the peripheral. However `self.did_update_value_for_characteristic`, which it schedules to be called by the event loop, is never being called. I also confirmed that `self._event_loop.is_running()` returns false when called at the end of `peripheral_didWriteValueForCharacteristic_error_`.

### What I Did

I have gotten my script to work by calling `self.did_update_value_for_characteristic` directly rather than scheduling it in the event loop, but that seems like it will eventually cause problems. Any suggestions would be welcome.

dlech, 2020-10-15:
Can you give a full working example to reproduce the bug?

dlech, 2020-10-15:
I suspect that the event loop that was captured when the delegate was created may not be the same event loop that the main program is run with.

dlech, 2021-06-03:
Please re-open if this is still a problem.

closed 2021-06-03 by dlech