hbldh/bleak #338: write_gatt_char() takes exactly 100 ms no matter what is the size of data

cshablin, 2020-10-19:
* bleak version: 0.8.0
* Python version: 3.8.5
* Operating System: Debian 10 (Buster)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.54

### Description

We try to pass 150 KB over BLE from RPI4 to NORDIC microcontroller via Raytac usb dongle.
Problem: no matter what is the size of packet to write (tried size 20 and 241) we get **exactly** 100 ms for each write operation.

I expect Write operation for 20 bytes each to be less than Write operation for 241 bytes each.
it is also looks suspicious that every write operation takes exactly 100 ms 

### What I Did

```
                        while bytes_sent < total_length:
                            to_send = min(self.config.max_ble_packet_size_bytes, total_length - bytes_sent)
                            content = data_packet.data[bytes_sent: bytes_sent + to_send]
                            await client.write_gatt_char(WRITE_CHAR_UUID, content)
                            bytes_sent += to_send
```

hbldh, 2020-10-20:
That is suspicious indeed. Does it happen when you write with Bleak to other devices as well?

dlech, 2020-10-22:
How did you measure the time?

cshablin, 2020-10-22:
> How did you measure the time?

added log(import logging) before and after method "write_gatt_char"

hbldh, 2020-10-22:
Do a

```python
import time

# ...

t = time.time()
client.write_gatt_char(...)
print(time.time() - t)
```

timing and verify that it prints out something close to 0.1 each time as well.

cshablin, 2020-10-28:
[timing_write.zip](https://github.com/hbldh/bleak/files/5452214/timing_write.zip)

attached log with timing as described:
t = time.time()
await client.write_gatt_char(WRITE_CHAR_UUID, content)
self.log.info("time for writing {} bytes is {} seconds", to_send, time.time() - t)

AGlass0fMilk, 2020-11-11:
@hbldh @cshablin 
I was also encountering this issue. I was trying to write a gatt char and writes were extremely slow. I had set it up so there was a timeout with asyncio:

```python
        t = time.time()
        await asyncio.wait_for(
            self._client.write_gatt_char(self.profile.write_char, data, response=with_response),
            timeout=0.01)
        print(time.time() - t)
```

My initial implementation had a basic timeout of 1 second. What I found was that the call to `await client.write_gatt_char` was _always_ timing out! So if there was no timeout it would just hang forever!

Not sure why this is happening. System info:

Output of `uname -a`: `Linux magic-man 5.4.0-52-generic #57-Ubuntu SMP Thu Oct 15 10:57:00 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

Output of `bluetoothctl --version`: `bluetoothctl: 5.53`

Output of `python --version`: `Python 3.9.0`

Output of `pip list`:
```
appdirs        1.4.4
attrs          20.3.0
Automat        20.2.0
bleak          0.9.1
constantly     15.1.0
hyperlink      20.0.1
idna           2.10
incremental    17.5.0
pip            20.2.4
PyHamcrest     2.0.2
pyserial       3.4
setuptools     50.3.2
six            1.15.0
Twisted        20.3.0
txdbus         1.1.2
zope.interface 5.2.0
```
Network interface/Bluetooth Chipset (`lspci`): `Network controller: Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter (rev 32)`

Other hardware info (`sudo dmidecode`):
Dell XPS 15 9570

Let me know if you need any other information. For now, adding a very short timeout using `asyncio.wait_for()` solves the problem as a workaround.

NbrTDB, 2020-11-16:
@AGlass0fMilk just for clarification, are you saying specifying a short timeout allowed the writes to *succeed* or does it just cause them to fail quicker if they weren't successful?

I'm dealing with a similar issue of slow writes (#357) and am looking for any clues I can! In my case, typically some ~1400ms for a single write with-response. 

Have you noticed that if traffic is continuous, the delay goes away?

cshablin, 2020-11-16:
> @AGlass0fMilk just for clarification, are you saying specifying a short timeout allowed the writes to _succeed_ or does it just cause them to fail quicker if they weren't successful?
>
> I'm dealing [...]

for me when traffic is continuous it goes up to exactly 100 ms each 'write' (see attached log of mine)
I don't think AGlass0fMilk's proposal will help us.

hbldh, 2020-11-24:
This looks some kind of connection interval problem in your peripheral, given that the times seems to be so consistent... Are you able to modify the connection interval on the peripheral yourself?

Example: https://stackoverflow.com/questions/51485129/uwp-bluetooth-le-api-is-slow

I also found this about modifying connection interval times on the Debian system you use to connect with: https://github.com/ukBaz/python-bluezero/issues/130#issuecomment-476003080

dlech, 2022-09-12:
There have been significant performance improvement made in the BlueZ backend in Bleak v0.17. So if this issue persists, it is likely something to do with the device itself.

closed 2022-09-12 by dlech