hbldh/bleak #342: corebluetooth returning "uuids" that are service names, like "Heart Rate Monitor" or "Device Information"

dhalbert, 2020-10-20:
* bleak version: 0.8.0 and tip of develop
* Python version: 3.8.2
* Operating System: macOS 10.15.7

I was startled to discover that corebluetooth appears to be returning English language strings for standard Bluetooth UUIDS, such as "Heart Rate Monitor".

In https://github.com/hbldh/bleak/blob/ccdac2388e882a5e5c446cf36df349fbfedce2e8/bleak/backends/corebluetooth/device.py#L45-L47
`cbuuids` can contain human-readable strings instead of strings that are UUIDs.

I was looking at the advertising data from a standard Heart Rate Monitor device (Scosche), and `cbuuids` contains this:
```python
(
    "Heart Rate",
    "Device Information",
    1725
)
```
bleak lowercases these, on the assumption they are uuids, but these are just readable strings.

So any attempt to determine if the advertised service is the Heart Rate Monitor service (0x180D) has to do a string comparison against "heart rate", instead of "0x180d".

I don't think bleak is doing this conversion, but it's coming directly from `kCBAdvDataServiceUUIDs`. I'm thinking `uuids.py` needs to contain a reverse dictionary to convert these strings back to hex for uniformity across platforms.

Am I missing something?? Has someone else encountered this? As usual Apple's documentation seems to be lacking details.

hbldh, 2020-10-20:
This is highly undesired. I have not experienced this myself, but will try to reproduce it at the end of the week. Does this happen "inside" the client as well, after connection, or is it only in the advertisement service uuids?

dhalbert, 2020-10-20:
It is what is returned in the advertisement data. I was trying to convert those strings to our own internal UUID objects (which parse the hex UUID strings), and got an error.

I think you would see this even on devices advertising Vendor IDs if they also provide Device Information Service, which most do.

PR #344 (2020-10-21): corebluetooth: use cb_uuid_to_str() to convert kCBAdvDataServiceUUIDs
This was missed in 6a0d8c99f5c82b6b1c000316fb1c0dff49a92138.

Also, since this was just using `str()` directly on the `CBUUID` instance, it had an additional bug for well known services since `str()` would return the name of the service instead of a UUID.

Fixes #342.
commit b980252c (2020-10-21): corebluetooth: use cb_uuid_to_str() to convert kCBAdvDataServiceUUIDs

This was missed in 6a0d8c99f5c82b6b1c000316fb1c0dff49a92138.

Also, since this was just using `str()` directly on the `CBUUID`
instance, it had an additional bug for well known services since `str()`
would return the name of the service instead of a UUID.

Fixes #342.
PR #348 (2020-10-22): Release/v0.9.1
## Added

* Added new attribute ``_device_info`` on ``BleakClientBlueZDBus``. Merges #347.
* Added Pull Request Template.

## Changed

* Updated instructions on how to contribute, file issues and make PRs.
* Updated ``AUTHORS.rst`` file with development team.

## Fixed

* Fix well-known services not converted to UUIDs in ``BLEDevice.metadata`` in CoreBluetooth backend. Fixes #342.
* Fix advertising data replaced instead of merged in scanner in CoreBluetooth backend. Merged #343.
* Fix CBCentralManager not properly waited for during initialization in some cases.
* Fix AttributeEr [... cut]
closed 2020-10-21 by dlech